Mr-MIBonk/M.I.B._More-Incredible-Bash #11: Can't download LFS files

kusmierz, 2022-06-13:
I can't pull any LFS file, as `git clone` raises an error: ` Smudge error: Error downloading 2022_03_MHI2_Password_List_V3.9.pdf`:

```
This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
```

harman-f, 2022-06-13:
This has to do with our (not optimal) git setup.

We had issues with github changing EOL on binary files and listed them as LFS.
Now we have this bandwith issue.

see .gitattributes

If you have an idea for a better config, you are welcome ;)

kusmierz, 2022-06-14:
@harman-f have a look at #12, I copied files from https://mib.mibsolution.one/ and turned off LFS in `.gitattributes`. For such small files it should be good enough anyway.

PR #12 (2022-06-14): #11 LFS fix
Fixes #11
commit 06d71fc6 (2022-06-14): Merge pull request #12 from kusmierz/lfs-fix

#11 LFS fix
closed 2022-06-14 by harman-f