Mr-MIBonk/M.I.B._More-Incredible-Bash #16: need help ！

yung56100, 2022-06-21:
I installed the SD card under the guidance. After loading the SD card (16macan), it shows that the next step cannot be carried out. Can you give me a hint of the proble
[image]
m

harman-f, 2022-06-21:
Are you using latest 3.3.3 version?

Format SD again and copy files newly.

yung56100, 2022-06-22:
The version I use is version 3.2.0. Is it caused by the mismatch between the version and the patch version ？

harman-f, 2022-06-22:
Try latest 3.3.3 version. 

And Format SD fresh before copying it over.

yung56100, 2022-06-23:
I tried according to your request. Now the Carplay function is enabled. It shows that it has been enabled, but it always shows that the USB connection is required. I checked the mobile phone and USB cable and found no problem.  thanks！！！
[image]

[image]
[image]

harman-f, 2022-06-23:
Please share your log file from /LOG/ on SD card. 

Also try different iPhone cable.

yung56100, 2022-06-23:
I have try different iPhone cable.[MU0654-AW7-MON16.02.1700061502.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/8966212/MU0654-AW7-MON16.02.1700061502.log)

harman-f, 2022-06-23:
Please share your backup. /backups/ on SD card. 
Your FW is not yet supported.

yung56100, 2022-06-24:
Do you have any other solutions? Thank you

harman-f, 2022-06-24:
No, backup needed to create patch. 

https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/wiki/Patch-missing%3F

harman-f, 2022-06-27:
@yung56100 

Issue can be closed?

yung56100, 2022-10-11:
OK
THANKS FOR YOU HELP

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2022年6月28日(星期二) 凌晨5:47
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [Mr-MIBonk/M.I.B._More-Incredible-Bash] need help ！ (Issue #16)

 
@yung56100
 
问题可以关闭吗？
 
—
直接回复此邮件，在 GitHub 上查看，或取消订阅。
你收到这个是因为你被提到了。消息 ID：<Mr-MIBonk/M 。我。乙。_More-Incredible-Bash/issues/16/1167944282 @ github 。com&gt;