Mr-MIBonk/M.I.B._More-Incredible-Bash #30: Patch for NavActiveIgnore.jar do not work in SEAT Harman 1447

jasoriab, 2022-08-26:
Hi

I have installed the latest version of the toolbox (4.3.1) in a SEAT León 5F, MIB2 Harman with the latest available firmware version, 1447

After installing the toolbox correctly, and making the backup, I have proceeded to activate the patch to ignore the active navigation in Android Auto, in order to continue seeing the map in the VC.

The patch is applied correctly, but after restarting the unit, the map in the virtual cockpit is still disabled, when activating a route in Waze or Google Maps.  If I try to apply the patch again, it warns that the file is already copied and does nothing.

Comparing the script that performs the modification, [https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/blob/main/apps/NavActiveIgnore](url) con el de mib2-toolbox [https://github.com/jilleb/mib2-toolbox/blob/master/Toolbox/scripts/install_nav_active_ignore.sh](url) 

I see that in the latter, mib2-toolbox modifies a script that loads the modified java library NavActiveIgnore.jar.

My question is, wouldn't it be necessary to do something similar here? In the logs, only this appears:

  RCC-NavActiveIgnore---->
  Reboot Unit...

  RCC-NavActiveIgnore---->
  Reboot Unit...

Thanks!

harman-f, 2022-08-26:
This patch was developed and tested on VW units.
Hence, Seat is most likel not compatable. 

The patching of lsd.sh, which is done by mib2-toolbox is not nessary, as the unit already loads all *.jar located in /mnt/app/eso/hmi/lsd/jars/ during boot.

I will exclude Seat from the supported devoces list and prevent installation on next M.I.B version.

harman-f, 2022-08-26:
originally developed here:
https://gitlab.com/alelec/mib2-lsd-patching

olli991, 2022-08-26:
Une the one in jille toolbox. I run it on 1447 and it works well :-)

jasoriab, 2022-08-26:
Hi @harman-f  thanks for your quick response and thanks for clarifying that it is not necessary to modify the .sh file to load the .jar libraries

But from what is exposed here, the patch is compatible with the SEAT Harman with firmware version 1447, [https://github.com/jilleb/mib2-toolbox/pull/189](url)

harman-f, 2022-08-26:
Your log output is a bit strange. On 2nd run you should have gotten a log entry stating, that the patch is already installed
[NavActiveIgnore.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9430776/NavActiveIgnore.txt)

remove .txt from the attached file and replace it in /apps/on your SD card. It will give a bit more log output
.

harman-f, 2022-08-26:
Will limit installation to the trains mentioned here: [https://github.com/jilleb/mib2-toolbox/pull/189](https://github.com/jilleb/mib2-toolbox/pull/189)

jasoriab, 2022-08-27:
> Your log output is a bit strange. On 2nd run you should have gotten a log entry stating, that the patch is already installed [...]

Ok, uptate the file NavActiveIgnore in the SD, and reflash de MIB tool in the unit.

These are the logs:

Logging_MU1447-MHI2_ER_SEG11_P4709_1-JTH-07604.07.193405003D:
HW Number: H55 - PN: 5F0035020J
Variant: FM2-H-ND-EU-SE-MQB
EEPROM Variant: MU-H-ND-EU
Dataset Version: V03935259E
Model ID: 047360 - Car Model: Seat Leon SC FR (5F5) 

 ls --> total 2919
      64 drwxrwxrwx 11 root      root          32768 Aug 26 13:23 .
      64 drwxrwxrwx 11 root      root          32768 Aug 26 13:23 ..
      64 drwxrwxrwx  2 root      root          32768 Aug 26 13:23 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Aug 03 17:31 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Aug 03 17:31 .gitattributes
      64 drwxrwxrwx  2 root      root          32768 Aug 03 17:31 .github
       1 -rwxrwxrwx  1 root      root            132 Aug 03 17:31 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Aug 03 17:31 2022_06_MHI2_Password_List_V4.0.pdf
       0 n---------  1 root      root              0 Aug 26 13:23 =MIB
      81 -rwxrwxrwx  1 root      root          41255 Aug 03 17:31 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Aug 03 17:31 LICENSE
      64 drwxrwxrwx  4 root      root          32768 Aug 03 17:31 Launcher
       1 -rwxrwxrwx  1 root      root             60 Aug 03 17:31 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Aug 03 17:31 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Aug 03 17:31 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6469 Aug 03 17:31 README.md
       1 -rwxrwxrwx  1 root      root             24 Aug 03 17:31 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
       0 -rwxrwxrwx  1 root      root              0 Aug 03 17:31 _Swdlautorun.txt
      64 drwxrwxrwx  3 root      root          32768 Aug 26 12:39 apps
      64 drwxrwxrwx  4 root      root          32768 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Aug 03 17:31 bash.png
      64 drwxrwxrwx  2 root      root          32768 Aug 03 17:31 config
      64 drwxrwxrwx  3 root      root          32768 Aug 03 17:31 esd
       4 -rwxrwxrwx  1 root      root           1995 Aug 03 17:31 metainfo2.txt
      64 drwxrwxrwx 10 root      root          32768 Aug 03 17:31 mod
      64 drwxrwxrwx  4 root      root          32768 Aug 26 12:38 patches
      16 -rwxrwxrwx  1 root      root           7989 Aug 03 17:31 start

RCC-backup---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1ad2097 (15933592)|93...215 (122):LOCKED
0x00ba0000 - 9820F300

Found JTH-07604.07.193405003D Backup, will check if all complete...
Version info..... -> done :-)
FEC folders...... -> done :-)
Shadow files..... -> done :-)
Dataset's........ -> done :-)
EEPROM........... -> done :-)
storage1......... -> done :-)
storage2......... -> done :-)
backing up backup -> Backup part 1 is already there!
ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1447
Free space on /net/mmx/mnt/boardbook 684M

RCC-launcher---->
-- _Swdlautorun.txt found
-- autostart already disabled
-- Skip MOST enabled
-- Illegal and withdrawn FECs removed
-- Cleanup done!
-- GEM version OK - no update needed
[... 33 lines cut ...]
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1447
Free space on /net/mmx/mnt/boardbook 684M

RCC-NavActiveIgnore---->
Enabled NavActiveIgnore
File already present on unit. Will be replaced
Reboot Unit...

-------

ls -alv /net/mmx/mnt/app/eso/hmi/lsd/jars/
total 548
drwxrwxrwx  2 root      root           1024 Jan 01  1970 .
drwxrwxrwx  8 root      root           1024 Jul 09  2019 ..
-rwxrwxrwx  1 root      root         250996 Jul 09  2019 GEM.jar
-rwxrwxrwx  1 root      root          23733 Aug 03 17:31 NavActiveIgnore.jar
-rwxrwxrwx  1 root      root           2923 Jul 09  2019 nullservice.jar

-------

NavActiveIgnore.jar is copied to the drive, so it appears in the MU1447-folders.txt file inside the backup folder, but unfortunately it still doesn't work, it always turns off the map rendering in the VC when I activate a route in Android Auto (I have tried with Waze and Google Maps).

This is the only modification I have made with M.I.B. Would it be necessary to modify any additional Android Auto options with the toolbox?

harman-f, 2022-08-27:
@olli991 
Are you also using AA or CP?

@jasoriab 
I just notice, that the script block to load any *.jar during boot does exist on Porsche, but not on Skoda.
Skoda/Seat/VW are normally same
Porsche/Bentley/Audi are also similar but different.

Thought this to be standard on all units...

lsd.sh
```
## Find and append jar files
##
if [[ -d $BASE_DIR/lsd/jars ]]; then
	JARS=$(find $BASE_DIR/lsd/jars/ -name '*.zip' -or -name '*.jar')
	if [[ ! "x$JARS" == "x" ]]; then
		for jar in $JARS; do
			BOOTCLASSPATH="$BOOTCLASSPATH:$jar"
		done
	fi
fi
```

however, runlsd.sh contains this:

```
for JAVA_CLASSPATH_DIR in $JAVA_CLASSPATH_DIRS; do
	if [ -d $JAVA_CLASSPATH_DIR ]; then
		JARS=$(ls $JAVA_CLASSPATH_DIR/*.jar)
		if [ "$JARS" != "" ]; then
			for jar in $JARS; do
				CLASSPATH="$CLASSPATH:$jar"
			done
		fi
	else
		if [ "x$JAVA_CLASSPATH_DIR" != "x " ]; then
			echo "Jar directory '$JAVA_CLASSPATH_DIR' not found"
		fi
	fi
done
```

This might also be a(nother) reason, why NavActiveIgnore is not working on Audi/Porsche/Bentley.
lsd.sh is different.
mibtoolbox is looking for $J9 string to be replaced, I think this will not work out as intended.
example from Audi lsd.sh:
```
if [[ -f $DEV_ACTIVATED && -f "$HMI_ZIP" ]]; then
	# start hmi.zip	
	info "starting j9 with $HMI_ZIP"
	$J9 $VMOPTIONS -Xbootclasspath:$BOOTCLASSPATH de.dreisoft.lsd.LSD & 
else
	if [[ -f "$LSD_JXE" ]]; then
		# start lsd.jxe
		BOOTCLASSPATH="$BOOTCLASSPATH:$LSD_JXE"
		info "starting j9 ...."
		$J9 $VMOPTIONS -Xbootclasspath:$BOOTCLASSPATH -jxe:$LSD_JXE &
	else
		echo !!!!!  lsd.jxe not found !!!!!
	fi
fi
```

Seems I have to look deaper into this.

olli991, 2022-08-27:
Carplay

harman-f, 2022-08-27:
any reports so far that AA is working on that FW?

harman-f, 2022-08-27:
@jasoriab
[NavActiveIgnore.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9438595/NavActiveIgnore.txt)

Please try this file

andrewleech, 2022-08-27:
Seat 1447 has been reported to work previously, 1407 was known to be incompatible.

The patch generally would work on all current versions of seat/Skoda/vw as they use compatible lsd.jxe Java code. 

It's known to not work on Audi as the lsd.jxe Java code is very different; different classes & functions for everything relating to the patch. I'm not sure but presume Porsche Java code will be more related to Audi than vw. 
If I get my hands on an Audi bench unit I could probably fix this, but don't know if that's on the cards.

jasoriab, 2022-08-28:
> @jasoriab [NavActiveIgnore.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9438595/NavActiveIgnore.txt)
> 
> Please try this file

Hi @harman-f 

I just tried the modified NavActiveIgnore file, and when applying the patch I get the following error (image attached) "Cannot find file /mnt/app/eso/hmi/lsd/lsd.sh"

So the problem continues, and the map in the Virtual Cockpit is deactivated when activating a route in Waze, for example.

[image]

Yesterday I gave Rainer the backup generated by M.I.B. in the SD before making any modifications to the unit, in case you want to take a look.

Thanks!!
Jose

harman-f, 2022-08-28:
[NavActiveIgnore.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9439258/NavActiveIgnore.txt)

try again, please

jasoriab, 2022-08-28:
Tried. Now it no longer gives the error that it can't find the file, and it patches it.

[image]

But unfortunately, it still doesn't work for me. The map in the VC continues to be deactivated when activating navigation in Android Auto.

jasoriab, 2022-08-28:
Hi @harman-f 

I have uploaded to the M.I.B_backup directory of mibsolution.one the last backup that M.I.B. toolbox generates, and I have included a subdirectory with the logs/ inside, in case they help you to diagnose what is happening.

Thanks again!
Jose

harman-f, 2022-08-29:
@jasoriab 
Finally found time to debug the script on my bench.
[NavActiveIgnore.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9442503/NavActiveIgnore.txt)

Please disable NavActioveIgnore 1st and than reinstall it.

Share LOG file and MUXXXX-lsd.sh from backup folder here.

jasoriab, 2022-08-29:
@harman-f a question. Is it necessary to re-flash M.I.B. on the drive, or changing the NavActiveIgnore file on the SD is enough? I think it is not necessary to flash again, but to be sure...

harman-f, 2022-08-29:
just rename the file - remobe .txt at the end an replace the one on sd in folder /apps/ no need to reinstall

jasoriab, 2022-08-29:
Uploaded both files compressed with gzip. The only modification made is to delete the last numbers of the VIN in the log file.

After applying it again, the patch still doesn't work. The map continues to be deactivated in the cockpit.

I hope that both files will help you to discover what is happening with my unit.

Thanks again!

[MU1447-JTH-07604.07.193405003D.log.gz](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9444255/MU1447-JTH-07604.07.193405003D.log.gz)
[MU1447-lsd.sh.gz](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9444260/MU1447-lsd.sh.gz)

harman-f, 2022-08-29:
Ok, LOG and lsd.sh look ok. All should be installed correctl now on your unit.

Still, not working?

jasoriab, 2022-08-30:
> Ok, LOG and lsd.sh look ok. All should be installed correctly now on your unit.
> 
> Still, not working?

No, unfortunately the patch still doesn't work on my unit. 

 I have tried Waze and Google Maps and when activating a route in them, the map in the Cockpit is deactivated.

andrewleech, 2022-08-30:
Maybe try uninstalling it then install with https://github.com/jilleb/mib2-toolbox/ instead to try to narrow down whether it's an issue with the patch itself, or the installation mechanism?

harman-f, 2022-08-30:
@jasoriab 
Installation method should be fine now.
Like suggested by @andrewleech, remove file with M.I.B and try toolbox.

jasoriab, 2022-08-30:
Ok, I have uninstalled the M.I.B. and I have tried mib2toolbox.

If I try to boot to the latest version on github, 4.11 I get this, without any options menu, and if I try to upgrade I get this error.

[image]
[image]

It also won't let me flash the latest version again. It is always marked as N in SWDL manual download.

[image]

If I try an older one, 4.0, the menus do appear, but within navigation, there is no patch for simultaneous navigation.

[image]
[image]

Anything else I can try? How to force mib2toolbox v4.12 installation again?

andrewleech, 2022-08-30:
You should be able to put an SD in with latest master download of toolbox and go to the update toolbox menu as seen on your second last screenshot

jasoriab, 2022-08-30:
Good news, I have followed the steps to force flash mib2tool as indicated here,

https://github.com/jilleb/mib2-toolbox/issues/202

And after being able to flash the latest version of mib2tool and run the update script, I managed to activate the patch successfully.

The VC map no longer disappears when activating a route in Waze, so for whatever reason the patch has been applied correctly this time.

Thank you very much to everyone for your help!!

harman-f, 2022-08-30:
Adjusted M.I.B script to resemble toolbox solution.
Will run some tests and included fixed version in next release

closed 2022-08-30 by harman-f