Mr-MIBonk/M.I.B._More-Incredible-Bash #31: Problems train to EU from USA

Suchar222, 2022-08-31:
Hello, i have problem train to EU from USA. 
My : MHI2_US_AUG11_S0533
[image]

Where is problem?

harman-f, 2022-09-04:
US FW is too old. binaries missing/not compatable. Update to latest US FW before conversion.

Suchar222, 2022-09-05:
And what update will work?

Suchar222, 2022-09-05:
MHI2_US_AU57X_K0539_MU1062 not work....

harman-f, 2022-09-08:
Yes, S version is not part of the supportedtrains list

```SupportedTrains = "MHI2_US_AUG11_P0526*,MHI2_US_AUG11_P0533*,MHI2_US_AUG11_P0534*,MHI2_US_AU57x_P0536*,MHI2_US_AU57x_K0538*"
```

harman-f, 2022-09-08:
[metainfo2.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9530062/metainfo2.txt)

Here you go!

In case you have a BOSE sound system installed, you have to either disbale it in update or remove it from metainfo.
Otherwise, you will need VCP/ODIS to upload new dataset.

Suchar222, 2022-09-09:
Thank You for metainfo but Your meta not work....
I send to the SD ->MHI2_US_AU57x_K0538_MU0109(photo)
I think problem is with software in my SD. 
I want final software MHI2_ER_AU57x_K3663_1_MU1425_IAO
Can You make my metainfo for this software:MHI2_ER_AU57x_K3663_1_MU1425_IAO
[image]

harman-f, 2022-09-09:
Your MMI is already converted to EU as it seem.
MHI2_US_AUG11_S0533 is still on?

Suchar222, 2022-09-11:
Yes , i think. 
This is my new photo:
[image]
[image]
[image]

harman-f, 2022-09-11:
Based on your screenshot:
[image]
Your MMI is set to EUROPE in coding.
And therfore installation is failing. Did you already try to convert it with some other means than M.I.B.
I did add Europe to supported region list.

Update to latest US FW. And use M.I.B from here to convert to EU.
 

[metainfo2.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9543533/metainfo2.txt)

Suchar222, 2022-09-12:
I was just convert only M.I.B

Suchar222, 2022-09-12:
I did update to MHI2_US_AU57X_K0538. (Done) 
Now I send to SD MHI2_ER_AU57X_K3663_1_MU1425_AIO
I have next problem.
[image]

Suchar222, 2022-09-12:
[image]

Suchar222, 2022-09-12:
[image]

Suchar222, 2022-09-12:
Software update *** Error A variant conflict occurred!  Main unit variant: MHI2_US_AU57x K0538 Version description variant: Train is blocked Update OPEN

harman-f, 2022-09-12:
You have to convert to EU 1st.

Uploaded image above are not working

Suchar222, 2022-09-13:
[image]

Suchar222, 2022-09-13:
https://ibb.co/Wy19Sr2

Suchar222, 2022-09-13:
If I have MHI2_US_AU57x K0538 i have problem with VCDS he cant connect module 5F... 
And i dont have green menu....

harman-f, 2022-09-14:
install M.I.B again to activate GEM
Convert region to EU
Inastall EU FW

Suchar222, 2022-09-15:
I installed M.I.B and i hold return+button (left down) 
Green menu does not work...
Why?
It happened when i installed it MHI2_US_AU57x_K0538_MU0109

Suchar222, 2022-09-15:
[image]

harman-f, 2022-09-15:
Sometimes getting into GEM can be a timing issue on pressing the buttons.
Try again, you can also try to install M.I.B again.

Suchar222, 2022-09-19:
I installed 10 times and nothing. Is there another way?

Suchar222, 2022-09-19:
Np. VCP?

harman-f, 2022-09-21:
GEM activation via M.I.B alsways workd ;)

Issue must have some other origin.

Suchar222, 2022-09-21:
I update to next software , i hope will by better.
Can You make my metainfo for this software MHI2_US_AU57x K0539 ? 
I have qestion? 😃
I can instal M.I.B by D-link ?

harman-f, 2022-09-22:
Which latest FW do you mean?
MHI2_US_AU57x_K3341_MU1426?

You can run all M.I.B function via putty from shell.

Suchar222, 2022-09-23:
No. i want MHI2_ER_AU57x_K3341_MU1426
For Europe. 
. 
I cant update to MHI2_ER_AU57x_K3341_MU1426
Why? Because not work M.I.B. 
What i have to did?
I had first time MHI2_US_AUG11_S0533
I have now MHI2_US_AU57x K0538. After update my module 5F not work. (No connect)
Mayby when i will update to MHI2_US_AU57x K0539 i hope "5F" will work.
 This is why i need metainfo for 
MHI2_US_AU57x K0539
I have:
-VCDS
-VCP
-D-link E100

justinthegeek, 2022-10-04:
> [metainfo2.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9530084/metainfo2.txt)
> 
> Here you go!
> 
> In case you have a BOSE sound system installed, you have to either disbale it in update or remove it from metainfo. Otherwise, you will need VCP/ODIS to upload new dataset.

How are you generating the metainfo files? I want to skip to k3324_1 but can't seem to find the trains along the way.

harman-f, 2022-10-08:
No tool, manual editing.
Which FW do you have currently on your unit?

MHI2_US_AU57x_K3324_1_MU0925
SupportedTrains = "MHI2_US_AU57x_P3237*,MHI2_US_AU57x_P3317*,MHI2_US_AU57x_K3324*"

justinthegeek, 2022-10-08:
I’m currently on MHI2_US_AU57x_K0538

justinthegeek, 2022-10-11:
I figured out how to modify the file, thank you for the push in the right direction :)

closed 2022-11-02 by harman-f

willblk973, 2022-12-30:
Hello, I have the same train MHI2_US_AUG11_S0533. Is it possible for you to create a meta file for me to go to MHI2_US_AU57x_K3341_MU1426

MrGonk, 2022-12-30:
> Hello, I have the same train MHI2_US_AUG11_S0533. Is it possible for you to create a meta file for me to go to MHI2_US_AU57x_K3341_MU1426

This is exactly what I need, too. Am I seeing this correctly, that the only alteration you need to make to the metainfo file would basically be to add "MHI2_US_AUG11*," to the "supportedtrains" list, written the same way all the other supported train references are?