Mr-MIBonk/M.I.B._More-Incredible-Bash #34: No input possible to screen after Patch

Pushkaaa, 2022-09-04:
Please help, No input possible to screen after patch...
after start it shows radio (about 1 sec) and stucks on Menu with no response...after some time shuts down...
With Dub-100 i can acces MMX only , so  cant restore ifsroot from backup...

harman-f, 2022-09-04:
Zip your log folder on SD and share here.

Pushkaaa, 2022-09-05:
HI, i think i made a fault, after a patch i made VIM install without reeboot...
its to big, cant upload everything...

[logs.zip](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9488206/logs.zip)

Pushkaaa, 2022-09-05:
> Zip your log folder on SD and share here.

[logs.zip](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9488996/logs.zip)

Pushkaaa, 2022-09-07:
Is there any possibility to restore a M.I.B backup on MMX conection with D-link? Or to run Emergency?
Or only way now is UART?

Pushkaaa, 2022-09-07:
I still can ascces M.I.B on SD through MMX console, but it dont aloud revert to Backup image...
Do i have any solution?Is there any possibility to restore a M.I.B backup on MMX conection with D-link?

harman-f, 2022-09-08:
flashing failed mid of erasing:
```
flashlock] unlocking flash: DONEFlashing on OFFSET 0x00ba0000...
Label: PL_qnx_flashit_10495A QNX: RL_qnx_650_10494A Built: 2010/12/10_11-21-47_UTC
/net/rcc/dev/fs0
--------
Flash size: 0x04000000 [= 512 x 0x20000]
Base addr:  0x00ba0000
File size:  0x00eea3a4 (15639460)
End addr:   0x01a8a3a4
*** erasing:     0x00BA0000 .. 0x01A8A3A3 [01A9FFFF] ..................................................................................................
```
How long do you have between reboots?

can you still access `/net/rcc/` from MMX?
If so show `ls -l` from `/net/rcc/dev`

closed 2022-09-12 by harman-f