Mr-MIBonk/M.I.B._More-Incredible-Bash #42: No patch for my unit?

skiv71, 2022-10-08:
Hi,

How can I proceed with patching my MIB2 Harmon when I cannot find a matching patch in the patches download?

Mine is MHI2_JP_VW270_P1001

Thanks

harman-f, 2022-10-08:
Create backup and share it here.

skiv71, 2022-10-08:
Ah, simple as that. Will do, thanks

skiv71, 2022-10-08:
I've done it... It's in MIB_Backup

skiv71, 2022-10-08:
Would it be possible to put EUR firmware on this thing?

harman-f, 2022-10-08:
Yes, EU conversion is possible.

skiv71, 2022-10-08:
Hmm... I'd definitely be interested in this... I've got experience with ISTA and E-SYS for BMW, but none with this VAG stuff..

Started doing some stuff, to get this JP import EU (UK) ready, but the maps was an issue.

Noticed I don't have DAB either...

harman-f, 2022-10-08:
Please open a new discussion about this.
So you do not need the patch right away?

skiv71, 2022-10-08:
ah, I will do... thanks.

Not sure what you mean "right away"?

I would like to get my MIB working properly asap, whether that's a patch for the JP or flashing a different f/w, I'm unsure.

skiv71, 2022-10-13:
Ok, so I'm running an EU f/w and it sort of works, but some things aren't right.

Can I restore my backup using MIB and it go back to stock?

harman-f, 2022-11-02:
What is not working for you?

skiv71, 2022-11-02:
Since changing the MHI2_JP_VW270_P1001 to MHI2_ER_VWG11_K3342...

a). I've lost the ability to configure Active Display Info (in the virtual cockpit dials) from the MIB2 unit.
b). I'm not able to show my navigation map in the virtual cockpit (between the dials).
c). During reverse, the MIB display randomly fails to switch into rear camera mode (and when it does, it sometimes doesn't switch out).
b). The menu graphics are at lot more dated.

I've found an identical unit in JP... I'm wondering if it would be possible to dump flash and restore that to my unit (so I can avoid Component Protection).

harman-f, 2022-11-06:
So your car came with stock AID/digital cluster?

The EU G11 FW does not support this.
Only MIB 2.5 --> G13 supports this stuff in EU FW.

Question now is which MIB display you have built into your car. 
MIB 2.5 needs the 9.2'' screen and does not support the older once.

As you have picture with your G11 FW now it seems, that you have 8'' display.

This is a mix not available like this in EU.

Can you share a picture of your screen and cluster?

Most likely you have to:
1. Get a 9.2'' screen
2 Convert to G13 FW

closed 2022-11-10 by harman-f