Mr-MIBonk/M.I.B._More-Incredible-Bash #53: New Patch required?

aaanimal, 2022-11-05:
Hi, thanks for all your work on this. I am trying to update to the latest ROW (New Zealand) maps on 2015 Audi TTS. I have activated MIB in the GEM, however there is no match for my current version:
MHI2_ER_AUG24_S1068
MU Software: 0364
I have uploaded my backup to mibsolution.one, could you provide the matching patch folder for me please?

harman-f, 2022-11-05:
Update FW to latest and patch that one.
mibsolution.one

harman-f, 2022-11-05:
Audi 2015... you will most likely also need to update the cluster.
Please check 1st which PN your cluster has:
https://mibwiki.one/doc/audi-visaidvckicluster-version-information-screen-3sDCZWyvjT

aaanimal, 2022-11-06:
Hi, that link doesn't work for me unfortunately. I signed up to Discord to access it and it still doesn't work.
What do you mean by the cluster? I'm not keen to update other ECUs other than the MMI

harman-f, 2022-11-06:
You must join the discord server 1st https://discord.gg/YU3rjatWjT or use telegram ;)

https://www.youtube.com/watch?v=fT_2lzO7Org

It depense on what you what want to achieve with patching.
For CarPlay/AA you also have to update the cluster, which is also working via SD card through your MIB unit.
All easy, if you have a cluster with index.

aaanimal, 2022-11-07:
Hi,
I don't know why, but that method doesn't seem to work on the TT Mk3.
Through the GEM I got the version info below:

APP: CLU3_MMX2_AU_ER_G24_038PROD
Branch: rel/CLU3
Changelist: 10352599
Framework (MMX): 5.31.8 MIB2CLU3 I118.2 CI47
Framework (RCC): 5.31.2 MIB2CLU3 I118.2 CI47
GraphicsServices: MIB2CLU3 93.1
J9: 1.08+ObjectSwapper_1.02+XIPLoader_1.10+MCFObjectReader_1.01C
DSI: 2_50_1
Navigation: CLU3_14411MIB2CLU3_B15.25.3.30203
NavDB: P39_N60S5MIBH3_ROW_NT_Q1.14
-----------
Navigation styles:
asia = 10.0.2198
aus = 10.0.2198
eu = 10.0.2198
il = 10.0.2198
india = 10.0.2198
meast = 10.0.2198
msa = 10.0.2198
msa2 = 10.0.2198
neast = 10.0.2198
za = 10.0.2198
-----------
EggnogDB: ROW_39_140627102
-----------
Truffles: 
Truffles.Version=376
Truffles.Variant=main_ROW
Truffles.Region=ROW
Truffles.Date=201407151322
Truffles.NavDB=NAVTEQ_2014_1
Truffles.UOTA_VERSION=560
-----------
Speech: 124.49.0-MIB2CLU3
Speech resources: 124.50.0
sseProc: 1.1.15
HMI: HMI_AU_MIB2_H_EU_G24_C3_150403
TextToolVersion:  6323
SDS-TextToolVersion:  6407
GCT version: 2.3.3.1639
VoiceEncoderApp: v2.1.102_FW5.31.8_TC17.24.05A
GracenoteDB: 7
Gracenote: Revision gdb-GL-0010-1.1.1.582-20141116 GNSDK4Auto.
Nvidia: 17.24.06-5639335
MME2 version: Media_1.0.45.3
exFAT-Library: 2013/11/26-22:47:12-MSK
Google Earth: v2.1.715_FW5.31.8_TC17.24.05B_v7_Bclu3
Streetview: v2.6.30_FW5.31.8_TC17.24.05B

IOC: 5304

RCC version info:
version  =  U7213
ipl-version = 1325082
ifs-emg-version = 941
ifs-root-version = 7718
efs-system-version = 0006552
efs-persist-version = 1999
dsp-version = 571811866

RCC label info:
label  = MIB2CLU3-Trunk_rcc2_B1-15282B

MMX Board information:
BOM: 699
Project: 61852
SKU: 0001
Version: 703
serial number: 0400415014960

Latest esolutions-changelists per depot:
mib2_bin: 10351968
mib2_com: 10331305
mib2_con: 10350791
mib2_diag: 10252570
mib2_epp: 10241877
mib2_fec: 10241877
mib2_ibr: 10279385
mib2_javatools: 10241877
mib2_mhd: 10287585
mib2_nav: 10316370
mib2_ols: 10301752
mib2_org: 10303132
mib2_rdv: 10249515
mib2_secure: 10241877
mib2_swdl: 10255461
mib2_tools: 10352617
mib_trf: 10259158
sysapp_spj: 10342954

Can that tell you what you need to know?

Ultimately I would like to update my navigation maps to a later version. If this is only possible via ODB or some other tool and Android Auto can be installed via the SD card, then that would be better for me.

harman-f, 2022-11-10:
These versions do not tell anything about the cluster. 
You have to check it by the method I linked you above.

Check cluster PN 1st

FW for MMI and Cluster - if it has index can be found on mibsolution.one

closed 2022-11-16 by harman-f