Mr-MIBonk/M.I.B._More-Incredible-Bash #57: Problem

Raptor0977, 2022-11-11:
Hi, I would also like to thank you for the great work which was done, share the problem and ask for advice.

I have got Q7, 2019.
Soft version MHI2Q_ER_AUG22_P5092 MU1329

My FEC is out of date - 0230002a, and I would like to replace it by 023000EE.
I have installed the paych without any errors, but nothing is happen. FEC is still the same.
Would you please advice what I did wrong and what I have to do with it?

Thank you in advance.

harman-f, 2022-11-11:
Please Share your LOG

Raptor0977, 2022-11-11:
[MU1329-JTB-07408.04.19117402AB.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/9991371/MU1329-JTB-07408.04.19117402AB.log)

Raptor0977, 2022-11-11:
__  __     ___           ___  
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - V3.3.3 "Bentley Edition - BYG24"
______________________________________________________________________

NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF

Logging_MU1329-MHI2Q_ER_AUG22_P5092-JTB-07408.04.19117402AB:
HW Number: 035 - PN: 4M1035036C
Variant: FMQ-P-TNL-EU-AU-MLE
EEPROM Variant: MU-P-LN-EU
Dataset Version: 4M0909545R
Model ID: 013706 - Car Model: Audi Q7 (4M0)

 ls --> total 2855
      64 drwxrwxrwx 10 root      root          32768 Jan 01  1970 .
      64 drwxrwxrwx 10 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  2 root      root          32768 Oct 03 19:26 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Jun 19 00:42 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            377 Jun 19 00:42 .gitattributes
       1 -rwxrwxrwx  1 root      root            126 Jun 19 00:42 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Jun 19 00:42 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Jun 19 00:42 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Jun 19 00:42 LICENSE
      64 drwxrwxrwx  4 root      root          32768 Jun 19 00:42 Launcher
       1 -rwxrwxrwx  1 root      root             60 Jun 19 00:42 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Jun 19 00:42 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Jun 19 00:42 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6238 Jun 19 00:42 README.md
       1 -rwxrwxrwx  1 root      root             33 Jun 19 00:42 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
      64 drwxrwxrwx  3 root      root          32768 Jun 19 00:42 apps
      64 drwxrwxrwx  4 root      root          32768 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Jun 19 00:42 bash.png
      64 drwxrwxrwx  2 root      root          32768 Jun 19 00:42 config
      64 drwxrwxrwx  3 root      root          32768 Jun 19 00:42 esd
       4 -rwxrwxrwx  1 root      root           1995 Jun 19 00:42 metainfo2.txt
      64 drwxrwxrwx 10 root      root          32768 Jun 19 00:42 mod
      64 drwxrwxrwx  4 root      root          32768 Jun 19 00:42 patches
      16 -rwxrwxrwx  1 root      root           7989 Jun 19 00:42 start

RCC-carp---->
CarPlay and Android Auto activated now

RCC-flash---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1ae05cb (15992268)|93...216 (123):LOCKED
0x00ba0000 - CC05F400

Found JTB-07408.04.19117402AB Backup, will check if all complete...
Partition info... -> done :-)
SWDL............. -> done :-)
Version info..... -> done :-)
Folder lists..... -> done :-)
FEC folders...... -> done :-)
HMI folders...... -> done :-)
Shadow files..... -> done :-)
lsd.jxe.......... -> done :-)
Dataset's........ -> done :-)
EEPROM........... -> done :-)
storage1......... -> done :-)
storage2......... -> done :-)
RCC Flashlock.... -> done :-)
RCC.............. -> done :-)
IFS-stage2.......4512+0 records in
4512+0 records out
 -> done :-)
OFFSET's......... -> done :-)
backing up backup -> Backup part 1 is already there! -> Backup part 2 is already there!

ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1329
Free space on /net/mmx/mnt/boardbook 816M

search File MU1329-ifs-root-part2-0x00ba0000-CC05F400.ifs

No Flashfiles there!

MMX-fecel---->

add FEC's to FEC File from backup folder
FEC File build done!

copy FEC file from /patches/MHI2Q_ER_AUG22_P5092_MU1329_PATCH... -> done :-)

MMX-fecel---->
copy EL file from /patches/MHI2Q_ER_AUG22_P5092_MU1329_PATCH... -> done :-)

MMX-reboot---->
Reboot Unit...

MMX-reboot---->
Reboot Unit...

harman-f, 2022-11-11:
Your log already shows the checksum of a patched ifs-root --> MU1329-ifs-root-part2-0x00ba0000-**CC05F400**.ifs

Please use latest M.I.B. It will also show the FECs on unit.

Raptor0977, 2022-11-11:
I have tried the last version the unit shows both FECs, but anyway everything is blocked :(

 ls --> total 2919
      64 drwxrwxrwx 11 root      root          32768 Jan 01  1970 .
      64 drwxrwxrwx 11 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  2 root      root          32768 Nov 11 22:37 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Oct 27 14:43 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Oct 27 14:43 .gitattributes
      64 drwxrwxrwx  2 root      root          32768 Oct 27 14:43 .github
       1 -rwxrwxrwx  1 root      root            132 Oct 27 14:43 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Oct 27 14:43 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Oct 27 14:43 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Oct 27 14:43 LICENSE
      64 drwxrwxrwx  4 root      root          32768 Oct 27 14:43 Launcher
       1 -rwxrwxrwx  1 root      root             60 Oct 27 14:43 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Oct 27 14:43 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Oct 27 14:43 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6469 Oct 27 14:43 README.md
       1 -rwxrwxrwx  1 root      root             21 Oct 27 14:43 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
       0 -rwxrwxrwx  1 root      root              0 Oct 27 14:43 _Swdlautorun.txt
      64 drwxrwxrwx  3 root      root          32768 Oct 27 14:43 apps
      64 drwxrwxrwx  4 root      root          32768 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Oct 27 14:43 bash.png
      64 drwxrwxrwx  2 root      root          32768 Oct 27 14:43 config
      64 drwxrwxrwx  3 root      root          32768 Oct 27 14:43 esd
       4 -rwxrwxrwx  1 root      root           1995 Oct 27 14:43 metainfo2.txt
      64 drwxrwxrwx 10 root      root          32768 Oct 27 14:43 mod
      64 drwxrwxrwx  4 root      root          32768 Oct 27 14:43 patches
      16 -rwxrwxrwx  1 root      root           7989 Oct 27 14:43 start

RCC-carp---->
CarPlay and Android Auto activated now

RCC-flash---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1ae05cb (15992268)|93...216 (123):LOCKED
0x00ba0000 - CC05F400

Found JTB-07408.04.19117402AB Backup, will check if all complete...
Partition info... -> done :-)
SWDL............. -> done :-)
Version info..... -> done :-)
GEM info........ -> done :-)
Folder lists..... -> done :-)
FEC folders...... -> done :-)
HMI folders...... -> done :-)
Shadow files..... -> done :-)
lsd.jxe.......... -> done :-)
Dataset's........ -> done :-)
EEPROM........... -> done :-)
storage1......... -> done :-)
storage2......... -> done :-)
RCC Flashlock.... -> done :-)
RCC.............. -> done :-)
IFS-stage2.......4512+0 records in
4512+0 records out
 -> done :-)
OFFSET's......... -> done :-)
backing up backup -> Backup part 1 is already there! -> Backup part 2 is already there!

ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1329
Free space on /net/mmx/mnt/boardbook 816M

search File MU1329-ifs-root-part2-0x00ba0000-CC05F400.ifs

No Flashfiles there!

MMX-fecel---->

add FECs to FEC file from M.I.B backup folder
Existing FecContainer parsed
no patch folder there, FEC File build is done in /tmp...
 -> done :-)
copy FEC file from backup JTB-07408.04.19117402AB... -> done :-)

MMX-fecel---->
copy EL file from  backup JTB-07408.04.19117402AB.... -> done :-)

RCC-addfec---->
FECs from /net/rcc/mnt/efs-persist/FEC
FECs currently used by unit
023000EE
02300044
00040100
0230002A
00050000
00070200
00030000
00060800
00060900

FECs from /net/rcc/mnt/efs-system/backup/FEC
FECs from factory
00040100
0230002A
00050000
00070200
00030000
00060800
00060900

MMX-reboot---->
Reboot Unit...

harman-f, 2022-11-11:
looks all fine:

RCC-addfec---->
FECs from /net/rcc/mnt/efs-persist/FEC
FECs currently used by unit
**023000EE**
02300044
00040100
0230002A
00050000
00070200
00030000
00060800
00060900

The "old" 023000XX will stay in list, the 023000EE will be relevant for map activation.
Should be working now.

Raptor0977, 2022-11-12:
I have spoken about this license expired
[image]
And here both FECS
[image]
Or I am wrong and this FEC did not linked to this license?

Thanks.

harman-f, 2022-11-12:
the licenses you are showing are for CONNECT.
No way to get these just by patching.

You have to get license from VAG

Raptor0977, 2022-11-12:
Ok.  Now everything is clear. Thank you.

closed 2022-11-12 by Raptor0977