Mr-MIBonk/M.I.B._More-Incredible-Bash #73: problems after I installed M.I.B. - More Incredible Bash

Vasi76, 2022-12-17:
Hello,

I need your help because I got 2 problems after installing "M.I.B._More-Incredible-Bash" on my machine (A4 B9 2016 MIB 2 High):
- Module 5F can no longer be accessed with OBDELEVEN - VCDS interfaces (unit is not responding)
- The HDD is no longer visible in the MMI unit

Thank you and I look forward to your help.

Regards,
Vasile

[MU1438-JTB-07429.10.1511850514.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10251510/MU1438-JTB-07429.10.1511850514.log)

 __  __     ___           ___  
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - V3.4.3 "More MOD..."
______________________________________________________________________

NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF

Logging_MU1438-MHI2_ER_AUG22_K3344-JTB-07429.10.1511850514:
HW Number: 050 - PN: 8W0035036A
Variant: FM2-P-TNL-EU-AU-MLE
EEPROM Variant: MU-P-LN-EU
Dataset Version: 8W0909545D
Model ID: 019401 - Car Model: Audi A4 B9 (8W0)
VIN -> CAR:  - MIB: 

 ls --> total 2567
      32 drwxrwxrwx 11 root      root          16384 Jan 01  1970 .
      32 drwxrwxrwx 11 root      root          16384 Jan 01  1970 ..
      32 drwxrwxrwx  2 root      root          16384 Nov 26 09:15 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Nov 16 14:21 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Nov 16 14:21 .gitattributes
      32 drwxrwxrwx  2 root      root          16384 Nov 26 10:50 .github
       1 -rwxrwxrwx  1 root      root            132 Nov 16 14:21 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Nov 16 14:21 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Nov 16 14:21 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Nov 16 14:21 LICENSE
      32 drwxrwxrwx  4 root      root          16384 Nov 26 10:50 Launcher
       1 -rwxrwxrwx  1 root      root             60 Nov 16 14:21 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Nov 16 14:21 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Nov 16 14:21 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6469 Nov 16 14:21 README.md
       1 -rwxrwxrwx  1 root      root             21 Nov 16 14:21 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
       0 -rwxrwxrwx  1 root      root              0 Nov 16 14:21 _Swdlautorun.txt
      32 drwxrwxrwx  3 root      root          16384 Nov 26 10:50 apps
      32 drwxrwxrwx  4 root      root          16384 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Nov 16 14:21 bash.png
      32 drwxrwxrwx  2 root      root          16384 Nov 26 10:50 config
      32 drwxrwxrwx  3 root      root          16384 Nov 26 10:50 esd
       4 -rwxrwxrwx  1 root      root           1995 Nov 16 14:21 metainfo2.txt
      32 drwxrwxrwx 10 root      root          16384 Nov 26 10:50 mod
      32 drwxrwxrwx  4 root      root          16384 Nov 26 10:51 patches
      16 -rwxrwxrwx  1 root      root           7989 Nov 16 14:21 start

RCC-backup---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1a89977 (15636856)|93...213 (120):LOCKED
0x00ba0000 - 7899EE00

Found JTB-07429.10.1511850514 Backup, will check if all complete...
Version info..... -> done :-)
FEC folders...... -> done :-)
Shadow files..... -> done :-)
Dataset's........ -> done :-)
EEPROM........... -> done :-)
storage1......... -> done :-)
[... 138 lines cut ...]
copy EL file from /patches/MHI2_ER_AUG22_K3344_MU1438_PATCH... -> done :-)
Reboot Unit...

RCC-NavActiveIgnore---->
no supported train found - will stop here

RCC-svm---->
Fix SVM Error (Note: The process can take some minutes...)
extract build version...
SvmInfo - Compressing... /net/rcc/mnt/efs-persist/SWDL/Log/SvmInfo.bin CfgAckRand - Uncompressing...
 /net/rcc/mnt/efs-persist/SWDL/Log/CfgAckRand.z to /net/rcc/mnt/efs-persist/SWDL/Log/CfgAckRand.bin

Key: 3A3D - CHKSum: F3EF

CfgAck - Compressing... /net/rcc/mnt/efs-persist/SWDL/Log/CfgAck.bin Wait for DSI to connect ...
DSI Connected role1.DSIE2PAccess
responseWriteE2P address 0x3f0 status 0 len 0x03
0x3f0 	00 00 01 
DSI Disconnected
clear SVM done!

Reboot Unit...

MMX-reboot---->
Reboot Unit...

harman-f, 2022-12-22:
Hi,

Module 5F can no longer be accessed with OBDELEVEN - VCDS interfaces (unit is not responding)
- Check your Train, is it still MHI2... or changed to MIB2...?

The HDD is no longer visible in the MMI unit
- There is no HDD built into your MMI unit. What do you mean by this?

Vasi76, 2022-12-22:
Hi,

Thanks for your answer.

The MMI unit has a hard disk drive used by Navi and a 10GB partition used for media files.
The navigation works fine, I can also see the size of the partition used by media files, but I can no longer access the files.

Regards,
Vasile

harman-f, 2022-12-22:
MHI2 does not have a HDD anymore. All stored on internal NAND.

Is there a fucntion to format the media partition?
Never have used it ;)

Maybe partition is damaged? Can happen to navdb, maybe also media

Vasi76, 2022-12-22:
Before installing M.I.B.,  all media files that I uploaded them were displayed in the MMI unit under "Media" and are now no longer visible (the internal memory is no longer visible). Format function is also no longer available.

Vasi76, 2022-12-22:
Could you please let me know if there is any possibility to install an internet browser in this unit?

Thanks.

harman-f, 2022-12-23:
> Before installing M.I.B., all media files that I uploaded them were displayed in the MMI unit under "Media" and are now no longer visible (the internal memory is no longer visible). Format function is also no longer available.

Nothing to do with M.I.B, as it is not touching anything related to that function.
Did you also update your FW, before patching?
Can you upload new media files to unit?

harman-f, 2022-12-23:
> Module 5F can no longer be accessed with OBDELEVEN - VCDS interfaces (unit is not responding)
> 
>     * Check your Train, is it still MHI2... or changed to MIB2...?
> 

Did you check this?

Vasi76, 2022-12-23:
Yes, the Train is MHI2

[image]

Vasi76, 2022-12-23:
if one of the 2 blue marked features is active then 5f module cannot be accessed by obdeleven

[image]

harman-f, 2022-12-23:
Normal, noting one should play around with. Will break stuff!

Vasi76, 2022-12-23:
if the feature checked in blue is active, then the media partition is invisible in the MMI.  unfortunately I found this out after I deleted it

[image]

harman-f, 2022-12-23:
Yes, you reduced your unit to kind of ony radio function ;)

[image]

Vasi76, 2022-12-23:
Do you know a way to install an internet browser on this device?
Thanks.

harman-f, 2022-12-23:
Nothing useable. 
The module displaying the board book is a simple browser.

closed 2022-12-26 by Vasi76