Mr-MIBonk/M.I.B._More-Incredible-Bash #75: Memory fault (core dumped) when trying to active Green Menu

johunb, 2022-12-20:
Firmware: MHI2_ER_AUG11_S0034_MU0073

I used M.I.B. via Putty because this has an on older GEM version.

I used Putty to connect via Telnet (MIB IP:172.16.250.248:123) and login as root. Root-Password for MHI2_ER_AUG11_S0034 is the same as for MHI2_ER_AUG11_P0040.

I inserted the SD card into slot SD1 of my MHI2 unit.
   # Mount SD card in slot SD1
     mount -uw /net/mmx/fs/sda0/
   # start M.I.B by typing
     /net/mmx/fs/sda0/start

Then selecting number 6: Developer Menu activation

I got: 
[image]

Memory fault (core dumped)
Memory fault (core dumped)
Developer menu now activated

Then the unit rebooted, but Green Menu was not activated. Tried it 3 times, but alwasy the same ... no activation of Green Menu.

Any ideas?

johunb, 2022-12-21:
Log-File:
 __  __     ___           ___  
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - 
RCC-dmenu---->
Developer menu now activated

Process 5193774 (pc) terminated SIGSEGV code=1 fltno=11 ip=01067960(/usr/lib/ldqnx.so.2@memcpy+0x228) mapaddr=00067960. ref=0000000a
Memory fault (core dumped) 

Process 5214254 (pc) terminated SIGSEGV code=1 fltno=11 ip=01067960(/usr/lib/ldqnx.so.2@memcpy+0x228) mapaddr=00067960. ref=0000000a
Memory fault (core dumped) 
Reboot Unit... press CTRL-C to abort!!!
Unit is Rebooting now...

RCC-dmenu---->
Developer menu now activated

Process 3731563 (pc) terminated SIGSEGV code=1 fltno=11 ip=01067960(/usr/lib/ldqnx.so.2@memcpy+0x228) mapaddr=00067960. ref=0000000a
Memory fault (core dumped) 

Process 3756139 (pc) terminated SIGSEGV code=1 fltno=11 ip=01067960(/usr/lib/ldqnx.so.2@memcpy+0x228) mapaddr=00067960. ref=0000000a
Memory fault (core dumped) 
Reboot Unit... press CTRL-C to abort!!!
Unit is Rebooting now...

RCC-dmenu---->
Developer menu now activated

Process 4571140 (pc) terminated SIGSEGV code=1 fltno=11 ip=01067960(/usr/lib/ldqnx.so.2@memcpy+0x228) mapaddr=00067960. ref=0000000a
Memory fault (core dumped) 

Process 4575236 (pc) terminated SIGSEGV code=1 fltno=11 ip=01067960(/usr/lib/ldqnx.so.2@memcpy+0x228) mapaddr=00067960. ref=0000000a
Memory fault (core dumped) 
Reboot Unit... press CTRL-C to abort!!!
Unit is Rebooting now...

harman-f, 2022-12-22:
Hi,

that FW version ist just too old and not supported by M.I.B.

Update your unit to MHI2_ER_AU57x_K3663_1
https://mibwiki.one/doc/mhi2_er_au57x_k3663_1-mu1425-aio-R6NuQH8h5p

closed 2022-12-29 by harman-f