Mr-MIBonk/M.I.B._More-Incredible-Bash #99: Make Subwoofer control visible on Audi B9 models with Audi Sound System

brucemiranda, 2023-01-29:
All the basic information is available in this toolkit to enable the hidden subwoofer menu in the Audi cars that have the Audi Sound System. e.g. B9 A4/A5 Q5 MKII A3 etc. 

You can in fact just reuse the amp app that is already written for the Canton. Except you need to remove the check for the Skoda train version. 

You just have to look at Module 5F at address 0x3B00 which is the same as the Canton app. 

If you find the value at that address to be 

004064F70009F70009F70009000000000000000000F700090000000002070102001E262E373F474F5764707D8995A6B6C7DBF03131CBBE 

you replace it with

004064F70009F70009F70009000000000000000000F70009F700090002070102001E262E373F474F5764707D8995A6B6C7DBF03131D568

And then reboot the MMI. The Subwoofer control will become visible in the MMI.

Mr-MIBonk, 2023-01-29:
Thank's a lot! I will build it in next days and tehn we can do more tests :-)

Silintone, 2023-01-31:
Im sorry but is there a txt file that shows a step guide , im not sure where to begin on this , what do i need so on .  i have a MIB2 unit with VC and nav ( 2016 databases :(  )

harman-f, 2023-01-31:
@brucemiranda 

I checked a few M.I.B backups and I could find that coding in quite a lot of them.

Could it be more generic?
Like all Audi cars with Sound System?

harman-f, 2023-01-31:
> Im sorry but is there a txt file that shows a step guide , im not sure where to begin on this , what do i need so on . i have a MIB2 unit with VC and nav ( 2016 databases :( )

Wait, there will be a function in M.I.B in the near future ;)

brucemiranda, 2023-02-01:
> @brucemiranda
> 
> I checked a few M.I.B backups and I could find that coding in quite a lot of them.
> 
> Could it be more generic? Like all Audi cars with Sound System?

It could well be more generic but I didn't know how much. I know my solution works on A3/A4/A5/Q5 MKII.
I think it's quite safe to assume that any car that has the address like I showed can be replaced with the new values given. The last 2 bytes are the checksum. So that would confirm that you are only updating cars that have exactly that value at that particular memory space. 
Ofcourse if you wanted to, you can always add a check if the car has the value '30' in byte 1 of the Module 5F at address 0x3000. That confirms the car has the Audi Sound System, which includes the Centre speaker and the Subwoofer.
I haven't seen what a car that just has the Base Stereo has at address 0x3B00 in Module.5F. I just know that the field is blank if the car has B&O because in that case the parameters are stored in the B&O amp module rather than the 5F MIB2.

harman-f, 2023-02-01:
We will compare the whole string included crc before replacing it.
Similar to the function for CANTON.

@Mr-MIBonk will look into an implementation in the near future. We will update you here (and ask you to run some tests ;) )

Silintone, 2023-02-02:
Hi Harman_f  /  Bruce .
i have the A.S.S , with center and sub , once you have your package compiled  i can be a test subject as , i need to update everything as well as after the sub control,  i do have a spare memory card available for this ,  let me know how i can help ( ps ill need a few "rookie how to steps" on how to do the test or checks
Thanks , Wayne

Silintone, 2023-02-05:
I managed to come right with installing the MIB, ive enabled vim .
The posted how to vid helped alot , ill upload my back soon,  just need to figure out what software to use to read and edit the sound oftion to be able to get to the 5f module Bruce mentioned

brucemiranda, 2023-02-05:
Thats my point. You won't need any additional software for my method. 5F is the MIB2's address and so any script in M.I.B. is already running on 5F. You just need to point at the right memory space, check if the subwoofer control is hidden and if so replace that memory space with the value that enables the subwoofer control. It doesn't get any easier than this. I only provided this method to this help this project since it's Free.

Silintone, 2023-02-05:
No problem , ill go dig some more in it now , Let me know how i can contribute , now that i figured out how to get it on the car.
More then willing to help :)   let me know here or via email.
PS thanks for the help thus far .

harman-f, 2023-02-05:
I will start to through some script together, as soon as something can be testet I will come back to you

Silintone, 2023-02-05:
Had a look , dont seem to be finding the option for the sub control( unsure of the menu path to follow ( apologies only had this software working for less then 2 hours)

Where can i upload my original backup before editing anything ? Please send the link to the upload page , not sure if that will come in usefull

harman-f, 2023-02-05:
@brucemiranda 
To be sure here, external amp it what we do NOT want to see in coding in this situation, right?
Byte 11 should be 1

harman-f, 2023-02-05:
@brucemiranda 
@Silintone 

this one was quick and dirty, no guarantee, that it will work - it might just do
https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/commit/96be10ec5a5c20d38b0bec60c2363bd1aa0ebec1
You have to download latest M.I.B from repo! Not one of the release versions.

Silintone, 2023-02-05:
@harman-f 

Brilliant , thank you ,  need to attend to a few things , then will test in about 2 hours from now ,  will advise on findings, need the logs even if it does work ?

harman-f, 2023-02-05:
LOGs and some pictures of the screen with/without subwoofer control would be nice ;)

Please also test both functions - to add and also remove it again!

harman-f, 2023-02-05:
get code from NOW.
Just adjusted some formatting, which might have caused an error.

brucemiranda, 2023-02-05:
> @brucemiranda To be sure here, external amp is what we do NOT want to see in coding in this situation, right? Byte 11 should be 1

I'm not sure about this. Might be worth checking a few backups for the value of Byte 11 when it has the DSNOSUB string I.e. Audi Sound System. But like I said, there are two sound configurations that do not use the external B&O amp. I.e. the Base Stereo and the Audi Sound System. I have no idea what that parameter dataset looks like in a car without the Audi Sound System I.e. the Base Stereo, which doesn't have the factory subwoofer.

brucemiranda, 2023-02-05:
> get code from NOW. Just adjusted some formatting, which might have caused an error.

Was just having a quick look at your code. Your else error message in -suboff is incorrect. 
It should say "No SUBWOOFER" rather than "No ORIGINAL" Dataset found.

Silintone, 2023-02-05:
Where can I upload to. Please give me link.
The patch didn't work,  stats no original dataset

harman-f, 2023-02-05:
> Where can I upload to. Please give me link. The patch didn't work, stats no original dataset Get Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

Just upload log file here.

Silintone, 2023-02-05:
[MU0358-JTB-07413.11.1511700D07.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10612007/MU0358-JTB-07413.11.1511700D07.log)

harman-f, 2023-02-05:
Yourt dataset does not match the one @brucemiranda posted.
I just pushed an update to repo.
LOG will now contain the dataset and long coding byte 11.

Please get that version and run again, so that we see your data in LOG.

brucemiranda, 2023-02-05:
> Your dataset does not match the one @brucemiranda posted. I just pushed an update to repo. LOG will now contain the dataset and long coding byte 11.
> 
> Please get that version and run again, so that we see your dataset and coding in LOG.

I was about to suggest that you get what the dataset looks like to begin with.

Silintone, 2023-02-05:
will test now , bit slow on how to work github so i just download the entire directory again in zip.  only way ive figure it out ,
Standby for results going to test now

brucemiranda, 2023-02-05:
> [MU0358-JTB-07413.11.1511700D07.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10612007/MU0358-JTB-07413.11.1511700D07.log)

@Silintone what car is this. Model and variant? And are you absolutely sure you have the Audi Sound System that comes with the Centre and Subwoofer from the factory, and not the Base Stereo?

harman-f, 2023-02-05:
Model ID: 000000 - Car Model: not set

This looks very wrong ;)

harman-f, 2023-02-05:
@Silintone 

please share the MU****-dataset.svc from M.I.B backup as well

Silintone, 2023-02-05:
Took these photos ig my car. I'll upload shortly 
[image]
[image]

Silintone, 2023-02-05:
[image]
[image]

Silintone, 2023-02-05:
2016  A4 B9  ,  CVK engine ,  140kw  FWB ,  south africa ( we done have that premium variants - everything is optional here ) 

[MU0358-datasets.csv
[MU0358-JTB-07413.11.1511700D07.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10612056/MU0358-JTB-07413.11.1511700D07.log)
](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10612054/MU0358-datasets.csv)

brucemiranda, 2023-02-05:
:-) @harman-f my string is upper case and what is being read is lowercase. No other issues. I'm sure you know what to do to fix this.

Silintone, 2023-02-05:
i know it must go lower case in the string , but where in the string i dont know , ill let Harman do it , i really dont want a expensive brick wired in my car

Silintone, 2023-02-05:
@brucemiranda , the first time you sent me a link there was a few photos of the person editing the cut off freq of the speakers and the gains ,  this is the next project id like to do, rear speakers are very dull and flat , no bass ,  even upgraded speakers are soft and bassless in the rear,  so i need to figure out how they did that , then i can maybe add that to this project .

harman-f, 2023-02-05:
> 2016 A4 B9 , CVK engine , 140kw FWD , south africa ( we done have that premium variants - everything is optional here )
> 
> [MU0358-datasets.csv [MU0358-JTB-07413.11.1511700D07.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10612056/MU0358-JTB-07413.11.1511700D07.log) ](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10612054/MU0358-datasets.csv)

Your long coding is NOT OK:
`00000009ff0a000013110001000a00102f0001e601700100de`

You di not have car model set, which should give you some issues...
Not sure what else might be wrong

harman-f, 2023-02-05:
> :-) @harman-f my string is upper case and what is being read is lowercase. No other issues. I'm sure you know what to do to fix this. I just checked and the original CANTON amp script also uses the lowercase hex notation. Just cover my string to lowercase case and let @Silintone test this again and it'll be fine.

Justed pushed the fix

Silintone, 2023-02-05:
i just saw that i have no main menu ,  when pressing the menu button and cycling from the car  through to audi and nav and phone ,  that screen stays grey ,   nav button remain at 11%

harman-f, 2023-02-05:
Take a picture...

This has some other cause, like your 000000 car model

Silintone, 2023-02-05:
How do I edit the car model etc?
Pics inc
[image]
[image]

harman-f, 2023-02-05:
M.I.B. -> System Information -> VariantInfo

brucemiranda, 2023-02-05:
@Silintone the subwoofer script should work for you now. Try that first and post the log again. 
But I have no idea what else is going on with your car. Have you played around with it before with VCDS or VCP or anything else? 
Try and hard reboot the MMI with the buttons and knob.

Silintone, 2023-02-05:
Good news is the menu and maps are back after changing variant info.
Sadly even both buttons up and pressing down the round knob , still no sub option.
Here are my logs 

[MU0358-datasets.csv](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10613781/MU0358-datasets.csv)
[MU0358-JTB-07413.11.1511700D07.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10613783/MU0358-JTB-07413.11.1511700D07.log)

Silintone, 2023-02-05:
Will need to continue tomorrow . need to call it for the night ,
Let me know if there is any uodate , or any other things you need me to test .
Thanks Bruce \ Harman

harman-f, 2023-02-05:
The last log shows, that the subwoofer dataset is installed on your unit.

Silintone, 2023-02-06:
Morning gents, I took this photo this morning.
I tried setting the varient to entry./high/premium option and no changes.
Tried the original thing of Changing car varient to a q7 but it doesn't appear.

Noting all tests are done and then a mmi reset with the 2 buttons forward while pressing round button down 

I'll be able to further later in the day 

[image]

Silintone, 2023-02-06:
Was playing with a few options during lunch time at work, tested with the 2 different A4 variants then installing the sub option   same result as before, no sub option with the grey screen instead of the main menu.

Had a look at other things, maybe this might older version maybe affecting it? 
See pics below. 

Is there ways of reading the backup files and editing them manually editing. 

[image]
[image]
[image]

harman-f, 2023-02-06:
> Morning gents, I took this photo this morning. I tried setting the varient to entry./high/premium option and no changes. Tried the original thing of Changing car varient to a q7 but it doesn't [...]

Never ever play with car model, or variant.
It does not have to do anything with what we want to du here and honestly you will only mess up your unit!
And make testing for us - who want to help you - impossible.

@brucemiranda 
Can you share a picture of how is should look like.
Could the old FW be an issue here?

brucemiranda, 2023-02-06:
There are potentially 3 issues that are affecting this test car. The FW is old, it's the older nVidia MIB2 and not the newer Qualcomm ones or the Subwoofer gain isn't set in address 0x3000

I'll need to find some time and look at the 0x3000 dataset from this car and check that first.

harman-f, 2023-02-06:
You mainly tested this on MHI2Q units?

If you give me more details I can have a look into M.I.B backups for the 0x3000.
Or also provide you some datasets from various backups.
Let me know what you need.

Is there already a documentation somewhere on how to read these datasets?

Silintone, 2023-02-06:
Did some homework , havent found out anything about nvidia / qualcom but i did see this ( i fall under the middle one ) 

[image]

brucemiranda, 2023-02-06:
@harman-f this is what my 0x3000 looks like. I had to make no adjustments to it whatsoever to get the subwoofer control to show up. 

**30**060E0050006F000700FA005A000700FA005A000700FA005A010E18380078**5A**0405003C2F323335363738393B3C3D3E4041424445466806071F400001141A20282D32373C3E4042444546484A680A0C0D0F1112141617191B1C1E2021232526280A0C0E10131517191B1E20222426282B2D2F310A0D101215181B1D202326292B2E313436393C0A0E1115181C1F23262A2D3134383B3F4246490A0E12171B1F23282C3034383D4145494E52560A0F14191E23282D32373C41464B50555A5F640A10151B21272C32383E43494F545A60666B71000A00960000000000000000000000000000000000000001C8012C0A141E1E1E1E1E282828323232323C3C3C3C3C080A00500000000000000000000000000000000000060F005FCD0000000000000000000000000000000006280140E600000000000000000000000000000000061F0258E5061303E8BF065E055ADF020A1F40E2002D00080A005000000000000000000000000000000000000611005BC500000000000000000000000000000000061900FFEC000000000000000000000000000000000618017CEF061704B0AE0625099DD202071B58E21E1700080A005000000000000000000000000000000000000611005BC500000000000000000000000000000000061900FFEC000000000000000000000000000000000618017CEF061704B0AE0625099DD202071B58E21E1700080700500000000000000000000000000000000000060F005FCD0000000000000000000000000000000006280140E600000000000000000000000000000000061F0258E5061303E8BF065E055ADF02071F40E2002D00080A001F0000000000000000000000000000000000090A008500000000000000000000000000000000000A0A00840000000000000000000000000000000000010A00C3DB061F0078DE060703E8000627003219410000080A01AE00000000000000000000000000000000000A0A173B000000000000000000000000000000000006150388A600000000000000000000000000000000063206CCCD063309C4D206231155CE06380280BEF60A00080A00500000000000000000000000000000000000060F005FCD00000000000000000000000000000000061E012CAD00000000000000000000000000000000061F0258E5061303E8BF065E055ADF020A1F40E2CE6400080A005000000000000000000000000000000000000625004A2800000000000000000000000000000000061900FFEC0000000000000000000000000000000006150096DA0617044CB80625099DD202071B58F11E17000807005000000000000000000000000000000000000625004A2800000000000000000000000000000000061900FFEC0000000000000000000000000000000006150096DA0617044CB80625099DD202071B58F11E1700080700500000000000000000000000000000000000060F005FCD00000000000000000000000000000000061E012CAD00000000000000000000000000000000061F0258E5061303E8BF065E055ADF02071F40E2CE6400080A001F0000000000000000000000000000000000090A008500000000000000000000000000000000000A0A00840000000000000000000000000000000000010A00C3DB061F0078DE060703E8000627003219410000080A01AE00000000000000000000000000000000000A0A173B000000000000000000000000000000000006150388A600000000000000000000000000000000063206CCCD063309C4D206231155CE06380280BEF60D00000000000100F6C800FA2800C800C82800C800C82800C800FA2800C800642800C801F4280036337C36

However byte 1 highlighted shows that the car has the Audi Sound System i.e. 6ch center sub (9.1)
And the other highlighted byte shows that the Subwoofer gain is set too. 
This means that the 0x3000 file is not a problem. 

Looking at the various datasets from the car we see that the dataset at 3221422112 (0xC0030020) was as we expected it to be. So we replaced with the dataset that makes the subwoofer control visible.  But that hasn't worked. So clearly we have another issue. Either the FW or the older nVidia units use another dataset for its controls.

Interestingly there is a dataset 3221422113 (0xC0030021) that is almost exactly the same as 3221422112 (0xC0030020) but in this dataset the Subwoofer is still turned OFF. Could that be the reason why @Silintone cannot see the Subwoofer control on his car? The only way to find out would be to change that too to make the Subwoofer control visible in that dataset too. Not sure if that will work but worth a try. If nothing else we just replace things as they were from the backup. 

So change 3221422113	(0xC0030021)

from

004060f70009f70009f70009000000f70009000000f700090000000002070000001e262e373f474f5764707d8995a6b6c7dbf02d2d77ec

to 

004060f70009f70009f70009000000f70009000000f70009f700090002070000001e262e373f474f5764707d8995a6b6c7dbf02d2d693a

But I am aware of some Audi A3 drivers that said they could only see the Subwoofer control once they upgraded the FW to the latest version. To perhaps that is the first time i,e. ensure the FW on this MIB is the latest before the datasets are updated.

brucemiranda, 2023-02-06:
@Silintone I think you should stop playing with your system and just concentrate on getting the latest FW for you MIB2 put on first. In the meanwhile @harman-f can process the information I have provided to see if we can figure out what might be going on here.

TheVaan, 2023-02-06:
A3 8V FL with ASS, FW MHI2_ER_AU37x_P4159 MU 1106:

Subwoofer control is visible but I don't hear an effect when playing around with it.

[MU1106-datasets.csv](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10663856/MU1106-datasets.csv)

Here is an excerpt of the log:

```MMX-subwoofer---->
Long coding byte 11:
1
Dataset found on unit:
004064f70009f70009f70009000000000000000000f700090000000002070102001e262e373f474f5764707d8995a6b6c7dbf03131cbbe
No external sound system coding found!
Install SUBWOOFER dataset
ORIGINAL dataset found on unit
Installing SUBWOOFER patch
DONE
Reboot Unit...```

brucemiranda, 2023-02-06:
> A3 8V FL with ASS, FW MHI2_ER_AU37x_P4159 MU 1106:
> 
> Subwoofer control is visible but I don't hear any differences when playing around with it.
> 
> [MU1106-datasets.csv](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10663856/MU1106-datasets.csv)
> 
> Here is an excerpt of the log:
> 
> ```
> Long coding byte 11:
> 1
> Dataset found on unit:
> 004064f70009f70009f70009000000000000000000f700090000000002070102001e262e373f474f5764707d8995a6b6c7dbf03131cbbe
> No external sound system coding found!
> Install SUBWOOFER dataset
> ORIGINAL dataset found on unit
> Installing SUBWOOFER patch
> DONE
> Reboot Unit...```
> ```

Glad to know it worked.

Trouble is that the subwoofer in the Audi Sound System is quite poor to conduct a test on. But try this, play a 40-50Hz test tone. This should only play on the subwoofer because the other speakers can't play this. Now increase and decrease the subwoofer and you should notice the subwoofer change its behaviour.

harman-f, 2023-02-06:
@brucemiranda 
I‘m not at my PC. How do the datasets on unit look in @TheVaan case?

brucemiranda, 2023-02-06:
> @brucemiranda I‘m not at my PC. How do the datasets on unit look in @TheVaan case?

They look exactly as I expected them, and tje fact that this enabled the Subwoofer for @TheVaan proves that we shouldn't look to update that other dataset that I was suggesting for @Silintone There is another issue that is affecting that car.

Silintone, 2023-02-06:
@brucemiranda , perfect ill look into updating it,  might just have to land up paying someone in europe ...... unless anyone is aware of a site i can download a update from

harman-f, 2023-02-06:
@Silintone 
Check M.I.B README!
There is a telegram channel. 
wibwiki.one
Mibsolution.one

harman-f, 2023-02-06:
You have an S version FW running on your unit. 
stock FW Update will most likely NOT work. 

I will have a look later and create a custom metainfo2 for you.

Silintone, 2023-02-06:
now that you mention it yeah , i see that mine is Sxxxx number 

For now im downloading
MHI2_ER_AUG22_K2161_MU1421 (A4-A5-Q5-Q7).7z
Compressed Archive MU Version - 1421

site is very slow apparently - unless the 30gb  maps file is killing the 100mb line , might as well do everything

[159 more replies]