Mr-MIBonk/M.I.B._More-Incredible-Bash #108: MIB2 Stuck on boot screen

NeNicYs7, 2023-02-11:
what happened
stuck on MIB2 VW boot screen

it was restarted and I can't open it anymore

remains off

please help...

harman-f, 2023-02-11:
I misplaced my crystal ball 😉 

please let me know what you did before. 
if you used M.I.B please share log file.

NeNicYs7, 2023-02-11:
So I installed mib2-toolbox, While checking what settings are available, I must have done something wrong because Discover Pro MIB2 is now stuck on the home screen.

The big problem is that I can't access any of the device menus, I can't go back to the green developer menu to undo what I've done, and I can't go into the service menu to flash the firmware and overwrite back to the default from works.

harman-f, 2023-02-11:
Log file is on SD Card in folder /Backup/ 
Please share that file here.

NeNicYs7, 2023-02-11:
but i don't have backup files

harman-f, 2023-02-11:
Log file is automatically generated and saved on SD card while using M.I.B. 

Basic Backup is generated during installation of M.I.B on unit. 

Just check your SD card!

NeNicYs7, 2023-02-11:
press the restart button, I don't even see anything on the black screen

how to do it, you have to repair the MIB2 unit ?

harman-f, 2023-02-11:
> installed mib2-toolbox

Ah, you installed mib2-toolboc. In that case you are wrong here.

I undersatnd, that your unit is not working. But without any usable information about what you did, there can not be any help.

Check out wibwiki.one for instructions on how to sue Telnet connection and on a broken unit better look for UART.