Mr-MIBonk/M.I.B._More-Incredible-Bash #111: MHI2 no AM and FM buttons in source

Jagorius, 2023-02-13:
Hi
I made a conversion (US->EU) from MHI2_US_POG11_P4151 to MHI2_ER_POG11_K5126. Everything ended well, navigation, android auto, sound (I have bose) works without a problem. The only problem I have is the lack of AM and FM selection buttons from the source tab
[image]

FM works fine, the List button shows a list of all available stations that can be selected. Another problem is that after pressing Tune button nothing happens, there is no window with the option to enter the frequency manually. Has anyone encountered such a problem? How to fix it? Should I try flashing only the tuner?

harman-f, 2023-02-13:
Did you use the conversion function in M.I.B GEM?
It should take care of all this.

M.I.B. -> Advanced Settings -> Convert to EU train

Jagorius, 2023-02-13:
Yes, i did

harman-f, 2023-02-13:
Please share your M.I.B backup and log

Jagorius, 2023-02-14:
Here is my M.I.B backup and log: https://mega.nz/file/zdhEyaqB#cAogPfaabPKR3npyg1NHZJm3IM1bWq7u8oAHKdVY3xc

harman-f, 2023-02-16:
I can not see in LOG, that you did run this on M.I.B GEM:
M.I.B. -> Advanced Settings -> 1-Click EU Coding

That will fix coding for EU

harman-f, 2023-02-17:
Have not seen that before and no idea right away what might cause it.
Looks like some coding issue.
Check manually via OBD if coding is OK now for radio region.

Did you had any issues during US to EU conversion?

Hardware issue?