Mr-MIBonk/M.I.B._More-Incredible-Bash #115: About patch for MHI2Q_CN_AUG22_P1404

johnzhouinfo, 2023-02-16:
Hi,
The MIB is working fine on this version, but I wonder if there be a patch for this version soon? 
ps: I already uploaded a backup to mibsolution.one in case you wish to look at it

harman-f, 2023-02-17:
You cane rename patch folder `MHI2Q_CN_AUG22_P1002_MU1002_PATCH` to `MHI2Q_CN_AUG22_P1404_MU1404_PATCH`

ifs-root-stage2 is the same between both versions

johnzhouinfo, 2023-02-18:
The patch was installed successfully. 
Thanks a lot!

closed 2023-02-18 by johnzhouinfo

toumahanabi, 2026-10-07:
> The patch was installed successfully. Thanks a lot!

Does the file `MU1002-ifs-root-part2-0x00ba0000-24D7F200.ifs` in that folder need to be renamed?

johnzhouinfo, 2026-11-06:
> > The patch was installed successfully. Thanks a lot!
> 
> Does the file `MU1002-ifs-root-part2-0x00ba0000-24D7F200.ifs` in that folder need to be renamed?

Sorry for the late reply! I guess you've probably figured it out by now. The script version I used back in 2023 generated the file directly as `MU1404-ifs-root-part2-0x00ba0000.ifs`.