Mr-MIBonk/M.I.B._More-Incredible-Bash #121: Activated carplay in mib

veicmario, 2023-02-20:
[image]

I have activated carplay but when i connect phone nothing happen. I get audi smarthphone option in menu and everything.

harman-f, 2023-02-20:
That is the AIO you installed. 

did you try to activate again with M.I.B —> log please. 

Have you checked if your USB port is comparable?

veicmario, 2023-02-20:
Hi, i sent you photos, i tried few times but no lick, two different cables and two phones but no luck.

If i heard right, i have no activation codes in red menu for carplay?

harman-f, 2023-02-20:
No photos received.

veicmario, 2023-02-20:
[image]
[image]
[image]

veicmario, 2023-02-20:
[MU1425-JTB-07414.03.17116600A6.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10787254/MU1425-JTB-07414.03.17116600A6.log)

harman-f, 2023-02-20:
And a picture of your USB port.

veicmario, 2023-02-20:
[image]

harman-f, 2023-02-20:
No phone symbol. 

Can you put a USB stick with mp3 files in and play them on unit?