Mr-MIBonk/M.I.B._More-Incredible-Bash #134: Add EQ controls for MHI2 Skoda

HeinanCA, 2023-02-27:
Hey,
I have Skoda Karoq with MIB 2.5 Retrofitted (Canton is original), SW version 1440.

I currently have 3 EQ fields, and I wonder if it is possible to increase this to 5 EQ sliders?
Meaning, also have Mid-Bass, Mid-Treble. Even with Canton, sound is not satisfying (moved from BOSE system on Audi A6 Avant which was far better).

Thank you!

harman-f, 2023-02-27:
Yes, it is possible, but the display is bugg in many GUIs on Skoda.

Get Notepad++
And open the subwoofer file from /apps/ on SD card

Change the marked `03` to `05`
[image]

And run the patch again in GEM.

Report back here with screenshots!

HeinanCA, 2023-03-01:
@harman-f  point is - I can't run the patch, because I have Canton, so it stops running (says "We will stop here, this is only for Internal sound system on 5F".

Anyway I can activate only the EQ sliders, with Canton?

harman-f, 2023-03-02:
That is a very good question.
No idea if it is enought to just change that setting to get the 2 additional working EQs.

Do you have VCP?
Doing it the manual way for testing would be quickist.
If you can confirm this to be working I could built soemthing around it.

HeinanCA, 2023-03-02:
No, only VCDS (unfortunately)

harman-f, 2023-03-02:
[subwoofer.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10870787/subwoofer.txt)

This is a dirty hack...
renaname the file back to `subwoofer` and replace in /apps/
run command in GEM again.

Report back.

HeinanCA, 2023-03-03:
> [subwoofer.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10870787/subwoofer.txt)
> 
> This is a dirty hack... rename the file back to `subwoofer` and replace the version in /apps/ run command in GEM again.
> 
> Report back.

Didn't work. Still 3 EQ only, not 5 😞

harman-f, 2023-03-03:
share your log from SD card.

HeinanCA, 2023-03-05:
[MU1440-JTH-07605.05.183402019C.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/10892246/MU1440-JTH-07605.05.183402019C.log)

Attached the logs :)

harman-f, 2023-03-07:
My hacked patched worked like it should.
In case of external amp the change in coding seems to be not effecting the controls.

This must be controlled by dataset in external amp.

HeinanCA, 2023-03-09:
Sorry for the noob question - can I somehow explore the dataset of the external AMP module?
Reminding I currently don't hold VCP..

harman-f, 2023-03-09:
I do not have any on hand

closed 2023-03-18 by harman-f