Mr-MIBonk/M.I.B._More-Incredible-Bash #143: Lost car's images

MuchaZ, 2023-03-12:
Hi,

After playing with MIB I've lost images of the car (eg. in the top menu and while parking). Any ideas?

Audi A5 Quattro
MHI2_ER_AUG22.K3346
1438

harman-f, 2023-03-12:
sounds like you messed araounf with coding...

M.I.B. -> Restore

`restore storage .raw - datasets from backup`

MuchaZ, 2023-03-12:
Thanks @harman-f for the reply. Is there a different way, if not having that backup?

harman-f, 2023-03-12:
Restore backup from NAND. 
You will find a new folder in Backup on SD card. 
copy all files one directory level up.

MuchaZ, 2023-03-13:
You mean "restore backup from unit to SD" from this menu, right?

[image]

harman-f, 2023-03-13:
[image]

closed 2023-03-18 by harman-f