Mr-MIBonk/M.I.B._More-Incredible-Bash #147: Activate audi drive select in audi a4 b9

Leor2208, 2023-03-16:
Hi team,
I would like to say thanks for your job, is amazing.
I have another challenge for you.
In audi a4 b9 with mhi2q exists cars that can select drive select mode (dynamic, sport, custom) and this selection can be done in mhi2 menú BUT you need to have a special physical buttons to use this function. Exist a coding to active this carisma menu but in mmi with the coding you cannot select the audi drive menu. Could you study the posibility of activate this menu?
Best regards!
Leo

Leor2208, 2023-03-16:
[image]

Leor2208, 2023-03-16:
This is the menu that we would like to activate in mih2q.
Maybe the name is carisma, maybe ADS or audi Drive Select.
By obdeleven you can activate the menus but it doesnt appear.
Thanks

Leor2208, 2023-03-17:
This is not a issue, sorry

harman-f, 2023-03-17:
the menu on MHI2Q is just some coding.

But there is more to be done outside of the 5F to get this going ;)

Leor2208, 2023-03-18:
We can do something?

harman-f, 2023-03-18:
If you can find out the required coding or adaptions

closed 2023-04-08 by harman-f