Mr-MIBonk/M.I.B._More-Incredible-Bash #158: Couldn't patch 4M Q7

angrycatmeowmeow, 2023-03-24:
Tried to patch_ifs_root_aio and it said it's only available for stock firmware. Everything worked good up until then.

https://drive.google.com/file/d/1TnWXPxRvsHlL3DrxK9XsHsWX_2KeKj1d/view?usp=sharing

Rohansinghdavid, 2023-03-24:
I'm in the same boat. I tried on my 2018 Q5 with software MHI2_US_AUG22_P3252. I reached out to @Mr-MIBonk  with the backup and log data but not received any response yet. Still waiting.

angrycatmeowmeow, 2023-03-24:
Yeah P3252 is where I'm at. Guess there's no patch for that FW. Can we update our own FW?

Rohansinghdavid, 2023-03-24:
I don't know. I'm waiting for experts like @Mr-MIBonk to reply and help me patch my system so that I can update to the latest maps.

Rohansinghdavid, 2023-03-24:
[image]
[image]
[image]

harman-f, 2023-03-25:
Well, so far there is just no patch for MHI2_US_AUG22_P3252 included in M.I.B.

Why not update to MHI2_US_AUG22_K3347_MU1439?

harman-f, 2023-03-25:
`SupportedTrains = "MHI2_US_AUG22_P3252*,MHI2_US_AUG22_K3336*,MHI2_US_AUG22_K3338*,MHI2_US_AUG22_K3340*,MHI2_US_AUG22_K3345*"
`

Rohansinghdavid, 2023-03-25:
@harman-f thanks for the reply. How do I update it to MHI2_US_AUG22_K3347_MU1439? Could you please help?

harman-f, 2023-03-25:
Your steps are correct. 

Mibsolution.one for files

Rohansinghdavid, 2023-03-25:
Thank you very much @harman-f . Will try it today and update you.

angrycatmeowmeow, 2023-03-25:
> Thank you very much @harman-f . Will try it today and update you.

Let me know how it goes. I may give it a shot too.

Rohansinghdavid, 2023-03-25:
Will do @angrycatmeowmeow . @harman-f just to let you know that I have B&O music system, do I need to take any precautions?

harman-f, 2023-03-25:
BOSE is an issue, I know that, not sure about B&O.

angrycatmeowmeow, 2023-03-25:
I have the Bose, which I guess requires a new meta file?

Rohansinghdavid, 2023-03-25:
Ok. So if something goes wrong with B&O, can is switch back to older software version and make the B&O work correctly? Is this an option?

Rohansinghdavid, 2023-03-26:
Hi @harman-f . I updated the firmware to MHI2_US_AUG22_K3347_MU1439 and then installed MIB but still not able to patch. It is taking the patch from the backup folder instead of the patch folder. I don't know why. Any suggestions?

Rohansinghdavid, 2023-03-26:
Hi @harman-f . I was able to patch finally. Thank you very much. It actually took the the patch MHI2_US_AUG22_K3345_MU1439 for firmware MHI2_US_AUG22_K3347_MU1439. Initially I had copied only one patch which are relevant. Later I copied all the patches to the SD and it automatically selected MHI2_US_AUG22_K3345_MU1439. I just have to update the maps now. Thank you very much again. Cheers!

Rohansinghdavid, 2023-03-27:
Hi @harman-f . I have an issue now. I tried the navignore to have the maps in VC while connected to Google maps via android auto. And as soon as I ran this the system is not booting at all. Please the attached video. It flickers and says initializing but later shuts off. I tried the MMI reset but again the flickering start for a while and then goes blank. Please help.  Thanks in advance. Rohan

harman-f, 2023-03-28:
Please Share your log file and a screenshot/picture of the output.

Rohansinghdavid, 2023-03-28:
Hi @harman-f . I created a new issue for my concern. Please see the below link.  
https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/issues/163
It has the log, backup and a video that I took.  
Please let me know if you need any further information.  

Thanks again for always responding to my concerns. 

Best regards. 
Rohan.

closed 2023-04-08 by harman-f