Mr-MIBonk/M.I.B._More-Incredible-Bash #163: MMI not booting after enabling dual navigation.

Rohansinghdavid, 2023-03-28:
I tried to enable the dual navigation on my 2018 Audi Q5. To complete the process MIB restarts the MMI and it never restarted.  The screen started to flicker and in VC "initializing" kept doing rounds. MMI shut off after like 1 minute. I tried MMI rest with left, right and center joystick button. It then started to repeat the same flickering and then stops and shows as "unknown" in the VC. 

I tried removing the fuse and disconnecting the battery terminals. Didn't resolve the issue. 

 __  __     ___           ___  
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - V3.5.0 "Spring Edition"
______________________________________________________________________

NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF

Logging_MU1439-MHI2_US_AUG22_K3345-YJL-26007.11.1679190015:
HW Number: 052 - PN: 8W0035880A
Variant: FM2-P-TNSL-US-AU-MLE
EEPROM Variant: MU-P-LNS-US
Dataset Version: 80B909545A
Model ID: 012406 - Car Model: Audi Q5 (8W5)
VIN -> CAR:  - MIB: 

 ls --> total 2919
      64 drwxrwxrwx 11 root      root          32768 Mar 25 22:04 .
      64 drwxrwxrwx 11 root      root          32768 Mar 25 22:04 ..
      64 drwxrwxrwx  2 root      root          32768 Mar 25 22:04 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Mar 12 09:08 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Mar 12 09:08 .gitattributes
      64 drwxrwxrwx  2 root      root          32768 Mar 12 09:08 .github
       1 -rwxrwxrwx  1 root      root            132 Mar 12 09:08 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Mar 12 09:08 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Mar 12 09:08 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Mar 12 09:08 LICENSE
      64 drwxrwxrwx  4 root      root          32768 Mar 12 09:08 Launcher
       1 -rwxrwxrwx  1 root      root             60 Mar 12 09:08 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Mar 12 09:08 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Mar 12 09:08 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6438 Mar 12 09:08 README.md
       0 -rwxrwxrwx  1 root      root              0 Mar 12 09:08 Swdlautorun.txt
       1 -rwxrwxrwx  1 root      root             24 Mar 12 09:08 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
      64 drwxrwxrwx  3 root      root          32768 Mar 12 09:08 apps
      64 drwxrwxrwx  4 root      root          32768 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Mar 12 09:08 bash.png
      64 drwxrwxrwx  2 root      root          32768 Mar 12 09:08 config
      64 drwxrwxrwx  3 root      root          32768 Mar 12 09:08 esd
       4 -rwxrwxrwx  1 root      root           1995 Mar 12 09:08 metainfo2.txt
      64 drwxrwxrwx 10 root      root          32768 Mar 12 09:08 mod
      64 drwxrwxrwx  4 root      root          32768 Mar 12 09:08 patches
      16 -rwxrwxrwx  1 root      root           7874 Mar 12 09:08 start

RCC-backup---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1a89977 (15636856)|93...213 (120):LOCKED
0x00ba0000 - 7899EE00

Found YJL-26007.11.1679190015 Backup, will check if all complete...
Version info..... -> done :-)
FEC folders...... -> done :-)
Shadow files..... -> done :-)
Dataset's........ -> done :-)
EEPROM........... -> done :-)
storage1......... -> done :-)
storage2......... -> done :-)
backing up backup -> Backup part 1 is already there!
ls -als /net/mmx/mnt/boardbook --> total 256
      64 drwxrwxrwx  4 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  4 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU0692
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1439
Free space on /net/mmx/mnt/boardbook 756M

[... 222 lines cut ...]
-rwxrwxrwx  1 root      root           4005 Jan 21  2019 xEventRecorderEvo.jar
-rwxrwxrwx  1 root      root          16749 Jan 21  2019 xTcpKeyPanel.jar
-rwxrwxrwx  1 root      root          44082 Jan 21  2019 xTcpLog.jar
-- Install NavActiveIgnore
navignore_audi.jar installed.
Unit will reboot now
Reboot Unit...

Background:
I had just updated from stock firmware MHI2_US_AUG22_P3252 to MHI2_US_AUG22_K3347 and then patched to update to latest maps just yesterday. While patching it took MHI2_US_AUG22_K3345 automatically. Everything was working fine. The navigation map were the latest and no issues were there. I just wanted to enable the dual navigation and this happened. 

Please please please please help. I am stressed out and panicking. 

I have shared the zipped backup and log folder in the below links. 
https://drive.google.com/file/d/1jY1QfEVTHtrzDdkQeBeDjnpe-2LNenor/view?usp=share_link
https://drive.google.com/file/d/14YOTFu1EqZt_dsY14g-1l_--gpzq79Fo/view?usp=share_link

I have also shared the video of the issue for your reference as the screenshots or images don't help much. 
https://drive.google.com/file/d/1E2sOvpbA4DYpbXcGMufi0yEK-8Qr8wkO/view?usp=share_link

Looking forward to your support in resolving this issue.
Thanks in advance. 

Best regards, 
Rohan.

Rohansinghdavid, 2023-03-28:
Yes, I tried to turn it off and on but no change. 

Also pulled fuse and disconnected the battery. Still not working. Does it mean that I bricked my unit? I connected the obd11 and it doesn't show the 5F control unit at all. There is error code on engine control unit as well which is DTC U105400 (Information electronics control module 1 No Communication). I'm not able erase this as well.  Hopefully there is a way out for me. Please suggest and help. Thanks in advance.

harman-f, 2023-03-28:
I do not see anything in M.I.B LOG, that should lead to a brick.
Patch was done one reboot earlier, so that has worked.

Did you leave some time between disconnecting and connecting power again?
Like min. 30s?
If you are too quick MMI will not power down fully.

Do you have D-LINK?

Next step would be to get UART equipment and check output on MMX RCC?

Rohansinghdavid, 2023-03-28:
No, I don't think I gave the min. 30secs after pulling the fuse or disconnecting the battery. Let me give that a try again this evening and update you. Fuse is mentioned as infotainment and not MMI. It is a 5amps fuse. Is that the correct one? And the battery in the boot needs to be disconnected and not the terminals under the hood in the engine bay, right? 

I don't have D-LINK. Nor heard about UART equipment. Will search for them on Google and get myself ready for it. 

I'll keep you posted.  

Thanks again for your timely support. Much appreciated.

Best regards.
Rohan.

harman-f, 2023-03-28:
MMI is important

Rohansinghdavid, 2023-03-28:
Ok. Will look for MMI fuse. And what is pull unit and quad lock? Is this the SD, sim and DVD unit? If how can you direct me to the instruction video or link. Else I'll search the Google if you just confirm. Thanks again.

harman-f, 2023-03-28:
Quadlock is the connector on the back of the MMI.
After you removed it from the car.

Rohansinghdavid, 2023-03-29:
Hi @harman-f . I did pull all the fuses which had "infotainment" mentioned in the description in the manual and left it for a couple of hours and then put back all the fuses and turned the ignition on. Voila! The MMI came back to life and loaded everything successfully. I bet I missed the right fuse during the 1st attempt like you said. Thanks for your advice. Much appreciate your ever supporting and helpful nature. You are the best! I was very much stressed and was panicking thinking that I might have bricked the MMI. Anyway, all good now. Do I need to insert the SD card again and take a new back up in the MIB? (I'm a bit scared now to run the MIB tool 😬.) Or just leave it as is? I think I need to connect the obd11 and clear all faults for sure.  Please let me know your suggestion and we can close this issue.  Best regards, Rohan.

closed 2023-03-29 by Rohansinghdavid

harman-f, 2023-04-08:
This can happen, but does happen rarely.
Use M.I.B as you wish. In case you get stuck again, pull known fuses ^^

Rohansinghdavid, 2023-04-08:
All done. Took the latest back up and everything is okay now. Thank you very much again @harman-f . Cheers!

kcisoul, 2024-02-29:
Which fuse did you pull out?? Too many infotainment fuse there.. 

> Hi @harman-f . I did pull all the fuses which had "infotainment" mentioned in the description in the manual and left it for a couple of hours and then put back all the fuses and turned the ignition [...]