Mr-MIBonk/M.I.B._More-Incredible-Bash #164: Android Auto not working after installation

keithpl, 2023-03-29:
I went through the installation process, rebooting the PCM several times. The PCM prompts if I want to use Android auto after connecting the phone and does nothing after I choose the Android Auto function.

It seems the IFS flashing step failed.

Log output:
```
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - V3.5.0 "Spring Edition"
______________________________________________________________________

NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF

Logging_MU1120-MHI2_US_POG11_P4173-JTB-07423.09.171184009C:
HW Number: 054 - PN: 9P1035880L
Variant: FM2-P-TNSL-US-PO-MLP
EEPROM Variant: MU-P-LNS-US
Dataset Version: -----------
Model ID: 077644 - Car Model: 
VIN -> CAR:  - MIB: 

 ls --> total 3039
      64 drwxrwxrwx 11 root      root          32768 Jan 01  1970 .
       0 drwxr-xr-x  2 root      root              0 Jan 01  1970 ..
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._.git-blame-ignore-revs
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._.gitattributes
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._.github
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._.gitignore
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._2022_06_MHI2_Password_List_V4.0.pdf
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._GEM.png
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._LICENSE
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._Launcher
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._PATCH COMPATIBILITY TABLE.pdf
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._README.md
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._Swdlautorun.txt
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._VERSION
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._apps
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._backup
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._bash.png
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._config
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._esd
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._metainfo2.txt
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._mod
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._patches
       8 -rwxrwxrwx  1 root      root           4096 Mar 27 02:27 ._start
      64 drwxrwxrwx  2 root      root          32768 Mar 27 02:24 .fseventsd
       2 -rwxrwxrwx  1 root      root            986 Mar 27 02:27 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Mar 27 02:27 .gitattributes
      64 drwxrwxrwx  2 root      root          32768 Mar 27 02:27 .github
       1 -rwxrwxrwx  1 root      root            132 Mar 27 02:27 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Mar 27 02:27 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Mar 27 02:27 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Mar 27 02:27 LICENSE
      64 drwxrwxrwx  4 root      root          32768 Mar 27 02:27 Launcher
       1 -rwxrwxrwx  1 root      root             60 Mar 27 02:27 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Mar 27 02:27 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Mar 27 02:27 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6438 Mar 27 02:27 README.md
       0 -rwxrwxrwx  1 root      root              0 Mar 27 02:27 Swdlautorun.txt
       1 -rwxrwxrwx  1 root      root             24 Mar 27 02:27 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  2098 WMPInfo.xml
      64 drwxrwxrwx  3 root      root          32768 Mar 27 02:27 apps
      64 drwxrwxrwx  4 root      root          32768 Jan 01  2098 backup
      77 -rwxrwxrwx  1 root      root          39351 Mar 27 02:27 bash.png
      64 drwxrwxrwx  2 root      root          32768 Mar 27 02:27 config
      64 drwxrwxrwx  3 root      root          32768 Mar 27 02:27 esd
       4 -rwxrwxrwx  1 root      root           1995 Mar 27 02:27 metainfo2.txt
      64 drwxrwxrwx 10 root      root          32768 Mar 27 02:27 mod
[... 177 lines cut ...]
00070200
00070100
00040100
06310099
00030000
00050000
00060700
00060800
00060100
00060A00

FECs from /net/rcc/mnt/efs-system/backup/FEC
FECs from factory
00060500
00070200
00070100
00040100
06310099
00030000
00050000
00060700
00060800
00060100
00060A00
```

harman-f, 2023-03-29:
00060900 is missing.

Have you changed your addFec.txt?

keithpl, 2023-03-30:
addFecs.txt in patches/MHI2_US_POG11_P4173_MU1120_PATCH ? No I have not, and when I look on the SD card 00060900 is there.

keithpl, 2023-03-30:
I went back and manually ran add_feccontainer.fec and now everything seems to be working well, not sure if I missed that in the instructions originally.

closed 2023-03-30 by keithpl

harlem1992, 2024-04-16:
Hi, Ive been trying to get my android auto working for the following patch MHI2_JP_POG11_K1503 but it connects and ask to allow for android auto but nothing happens after that. Apple Carplay works seemlessly for this before intallling MIB.

I saw an option to convert to EU, should i do that or better to just leave it as it is.