Mr-MIBonk/M.I.B._More-Incredible-Bash #169: VIM

errornt, 2023-04-05:
I have problem with toolbox 3.5.0. 

CarPlay and FEC was okay I made maps update, but VIM doesn’t work. 
it says all done, system reboot. 

But after trying it, still same,  6kmh. 
[image]

errornt, 2023-04-05:
[image]

harman-f, 2023-04-05:
Well known bug in FW 😢

errornt, 2023-04-05:
> 

Solution? I have plans to try with older version of Toolbox. 
Should work?

harman-f, 2023-04-05:
No, will not help!

issue is not related to M.I.B. 
the FW itself has this bug, that changed to VIM do not have any effect.

errornt, 2023-04-05:
> No, will not help!
> 
> issue is not related to M.I.B. the FW itself has this bug, that changed to VIM do not have any effect.

Understand. 
Thx for info. 🍻

closed 2023-04-08 by harman-f