Mr-MIBonk/M.I.B._More-Incredible-Bash #170: ABOUT PATCH

INH888, 2023-04-06:
Patch folder needs to add new version,MHI2_CN_VW48x_P1102_MU1102 needs to be supported...
PLEASE PLEASE my G.O.A.T

INH888, 2023-04-06:
Or do you have the 7720 version of China map?

INH888, 2023-04-06:
Or do you have a map of China with the map version code 7720? Thank you very much!

harman-f, 2023-04-06:
Upload your backup to mibsolution.one
You can also check maps there.

INH888, 2023-04-06:
> Upload your backup to mibsolution.one You can also check maps there.

Unfortunately, I have tried many times, but the M.I.B.3.5 program does not work successfully on the 1102 version of the MIB048. What should I do?

harman-f, 2023-04-06:
Please share your M.I.B log and a picture of the error.

INH888, 2023-04-06:
> Please share your M.I.B log and a picture of the error.

Okay, I'm being stupid. I've uploaded what you need to mibsolution.one

INH888, 2023-04-08:
> Please share your M.I.B log and a picture of the error.

Dude, any news on patch with firmware version 1102? I want to update the fsc code...Please forgive my poor English. 
Best wishes

harman-f, 2023-04-08:
Just checked your backup and LOG.
Where do you get stuck?

LOG lokks OK, but does not show any patch attempt!
Backup is only the one from M.I.B install, which does not contain all files.

Run the AIO patch and see what happens. A patch for your FW is already available.
Have you added the patch files to your SD?

https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/wiki/Prepare-M.I.B-SD-card-&-install-M.I.B#add-patch-files
[image]

INH888, 2023-04-08:
> 

> Just checked your backup and LOG. Where do you get stuck?
>
> LOG lokks OK, but does not show any patch attempt! Backup is only the one from M.I.B install, which does not contain all files.
>
> Run the [...]

This is the problem, there is no matching software version in this AIO PATCH.Can I use the 0841 version of AIO file[image]

harman-f, 2023-04-08:
Check patch archive on mibsolution.one in a bit.
Patch should be there.

INH888, 2023-04-08:
> 稍后检查mibsolution.one上的补丁存档。补丁应该在那里。

[image]
Sorry, it was my fault earlier, wasting your precious time, not reading all the instructions on M.I.B. Now, I have uploaded the firmware backed up by m.i.b to the m.i.b_backup folder again, please help me make the patch, thank you very much again, please forgive my bad English, my good brother.

INH888, 2023-04-08:
> 稍后检查mibsolution.one上的补丁存档。补丁应该在那里。

Thank you! The AIO patch is working! Love you!

closed 2023-04-09 by harman-f