Mr-MIBonk/M.I.B._More-Incredible-Bash #172: Unexpected Problems after installation

bulerd, 2023-04-09:
**Describe the bug**
Hi,  I have 2016 A4 2.0 TDI Sport with Virtual Cockpit Display(stock) and in order to update Navigation Maps for Turkey  have installed MHI2_ER_AUG22_P3241_MU0678 and GEM 4.5 installed with M.I.B_patches_2023_04_08. Before installation MHI2 version was MHI2_ER_AUG22_P2148.  At the end of the day, map couldn’t be updated (P101_N60S5MIBH3_ROW from MIB Solutions) despite adding 023000ee, car setting menu in MHI2 disabled and Navigation, Telephone and Media tabs are disabled. (Tabs are present but infos seen on MIH2 including map aren’t shown in Virtual Cockpit Display. 

I've checked every line of your page about this issue, and after several attempts I did right installation. While attempts i couldn't install  anything, so didn't change anything in GEM.

After installation of GEM 4.5, first I got back  and just patched  FEC and MHI2_ER_AUG22_P3241_MU0678 from M.I.B_patches_2023_04_08.

[MU0678-JTB-07423.04.161183030B.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11185738/MU0678-JTB-07423.04.161183030B.log)

[image]
[image]

Thanks fpr you help..

harman-f, 2023-04-09:
Well...

Turkish maps is most likely a conversion from EU to ROW.

You need to change coding and also ROW map FEC 023**D**00ee.
Change addFec.txt and run addFec again from GEM.
For coding M.I.B does not have an automatic function.
You need to change maps region (decimal 10) and SDS, as well as some stuff for radio...

yalmac, 2023-04-09:
Contact with me can fix it in a min. No problem. You can talk also Turkish :)
Telegram : https://t.me/yalmac
But; as @harman-f mentioned you need correct fec codes.

bulerd, 2023-04-09:
Thanx, i will try to do that but other problem has become more important for me. What should i do for fixing Car setting and Virtual Cockpit display problem?  :((((

harman-f, 2023-04-10:
Could you fix your issue(s)?

bulerd, 2023-04-10:
Firstly, very much appreciation to @yalmac . He found the problem that caused  while restoring from backup. In dataset.csv file, it was reset to zero value in sum lines that's why virtual cockpit couldn't get maps and other infos from MIB. Unfortunately i have no original datasets of my car. @yalmac re-coded zero lines with another a4's dataset. But the thing is that: My car is 2016 2.0 TDI Sport the other car is 2016 TFSI Design with almost same accesories except drive select modes. I have four the other has 3 modes. I will try to restore backup from dataset that @yalmac recoded tonight. I hope it solves, if it doesn't i will try online coding.

harman-f, 2023-04-10:
M.I.B is not touching the datasets on unit.
Something must have happend to reset them to 0...
If this happend after M.I.B installe, the backup created during M.I.B installation does hold your original coding!
No need to get datasets froma different car.
In case you deleted that backup from SD, a copy is stored on your unit and can be recovered back to SD card!

In the pictures you compare your initial backup (left) with a new one (right)?
Or yours (left) and Yalmac (right)?

bulerd, 2023-04-10:
Recoded one on the left. I will apply  to @yalmac  what you wrote about recovering copy back to SD card.

bulerd, 2023-04-10:
Before installing toolbox mentioned in my first comment, i had installed another toolbox (i have mentioned in #171 ), Audi-MIB2-Toolbox-master. But after installation because i couldn't patch the FEC (i made 2-3 attempts to do it just by inserting sd card and waiting to be installed, surfing in menu's without cahnging any configuration), I found toolbox (mentioned above), uninstall the first one and installed new one that i achieved to patch eventually.   While in my first installation, ı dumped everything to SD card, and i still have it. But i don't know if it is usable to solve my problem :(

bulerd, 2023-04-11:
Caused by me and solved by you. Thank you very much @yalmac @harman-f

bulerd, 2023-04-11:
by adaptation via OBD, it is solved.

closed 2023-04-11 by bulerd