Mr-MIBonk/M.I.B._More-Incredible-Bash #176: DUEL NAVI

INH888, 2023-04-13:
Dude, I have a new problem, there is a problem with the dual navigator, as shown in the image ...

[image]
[image]

harman-f, 2023-04-13:
Need to see your train. 
best upload log file.

INH888, 2023-04-14:
> Need to see your train. best upload log file.

[MU1102-AW7-MON22.06.1700062600.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11228960/MU1102-AW7-MON22.06.1700062600.log)

harman-f, 2023-04-14:
Get latest version from repo: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/archive/refs/heads/main.zip
And report back

INH888, 2023-04-15:
> Get latest version from repo: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/archive/refs/heads/main.zip And report back

[MU1102-AW7-MON22.06.1700062600.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11238756/MU1102-AW7-MON22.06.1700062600.log)

harman-f, 2023-04-15:
get fresh version, should work now

INH888, 2023-04-16:
> get fresh version, should work now

Still not working T.T
[MU1102-AW7-MON22.06.1700062600.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11241186/MU1102-AW7-MON22.06.1700062600.log)

INH888, 2023-04-19:
> get fresh version, should work now

Master，Have new progress?

closed 2023-04-20 by INH888