Mr-MIBonk/M.I.B._More-Incredible-Bash #184: Installation is successful but M.I.B. - More Incredible Bash does not appear in GEM

neoalius, 2023-04-23:
Installation is successful but M.I.B. - More Incredible Bash does not appear in GEM

BACKUP/LOG FOLDER is empty
[image]

harman-f, 2023-04-23:
Which FW?
Did you leave SD in unit?

benassyGT, 2023-04-23:
Same issu like u.. on audi

neoalius, 2023-04-23:
> Which FW? Did you leave SD in unit?

Yes ! SD card was in the card slot while final restart

firmware MHI2_ER_AUG_K3346 MU 1438

harman-f, 2023-04-24:
please share log file

benassyGT, 2023-04-24:
How?? If i not run M.I.B???

neoalius, 2023-04-26:
BACKUP/LOG FOLDER is empty

benassyGT, 2023-04-30:
> BACKUP/LOG FOLDER je prázdná

Yes, its empty.. mayby try instaling another FW

neoalius, 2023-04-30:
Issue fixed

Reset the unit to factory except media
Used a different SD card adapter ( made sure it’s not locked) and updated MMI again. Wolla it worked.

closed 2023-04-30 by neoalius