Mr-MIBonk/M.I.B._More-Incredible-Bash #200: MAP update ERROR : No valid update data

neoalius, 2023-05-07:
**Describe the bug**

Unable to update MAPs, ERROR : No valid update data. 
Unit Patched with FEC 
S/W : MHI2_ER_AUG_K3346_MU1438

**Add M.I.B LOG file to your report**
[MU1438-JTB-07428.05.16119300F9.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11413667/MU1438-JTB-07428.05.16119300F9.log)

**Screenshots**
[image]
[image]
[image]
[image]

**Additional context**
Tried updated in REM
Reads NAV update but not possible to start update.
[image]
[image]

Country is Australia and genuine NAV DB FEC is 023D0020 

SD cards is 64 Gb exFAT format 
[image]

neoalius, 2023-05-07:
Update started after replacing 023000EE with 023D00EE

closed 2023-05-07 by neoalius

harman-f, 2023-05-07:
SDS also has to be switched. 

FEC and coding have to be all correct to allow map install. 
unit was EU before?
You might need to delete maps on unit. 
there is a function in M.I.B