Mr-MIBonk/M.I.B._More-Incredible-Bash #206: Final Script Take to Long to Execute

R3DcooDe, 2023-05-10:
Hello

Audi A6 2016 Imported from Korea
It has MHI2_KR_AU57x_S0035

I try to install M.I.B via sd Card ( Note i this problem occurs many time i use in 4 diferent cars Audi A6 same version)  with diferent sd cards new ones

Problem is when intry to install mib,  unit stuck on final script for long time. After it shows an errorr se photo below.   After i press try again unit restart after some times. At final screen it says LANUCHER OK.  but when i enter green menu there is no MIB,  

So i have to reapet Installling procedure Again.  Same steps red menu sd card only launcher set to yes. Pres update. Wait long time at final script.   After restart mib is installed 
[image]

[image]

harman-f, 2023-05-10:
This is a very old FW ;)

Try latest version from repo: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/archive/refs/heads/main.zip
It just added some initial support for these old FW. Not all M.I.B functions are working, but the most basic stuff might now ;)

Please run it and share the log file from SD

R3DcooDe, 2023-05-10:
Hello Mate, thank you for your great work,  here in my country all cars almost are with old firmwares :/  i use mib to convert to Europe version, and install European Firmware.  
Here are logs for today region Convert. Korean firmware and European Firmware 

[MU0035-AW7-MON05.10.1500060674.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11444357/MU0035-AW7-MON05.10.1500060674.log) -- Korea

[MU1425-AW7-MON05.10.1500060674.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11444359/MU1425-AW7-MON05.10.1500060674.log) -- Europe

harman-f, 2023-05-10:
So it did work in the end?

R3DcooDe, 2023-05-10:
yess well done sir well done, keep up the great work.   if you have some time, can i contact you somewhere else, to discus about mib2+ mib3 solutions :D  i have telegram Tahyp Lala and email address r3dcoode@outlook.com   thank you :)

closed 2023-05-12 by harman-f