Mr-MIBonk/M.I.B._More-Incredible-Bash #212: Need help on patching ifs-root

zzz08900, 2023-05-16:
I was trying to add FEC to my MHI2 unit but so far no luck. My MHI2 unit seems to  be a rather strange unit with the following SW Train: MHI2_CN_VW511_P1002. 
I managed to install M.I.B and applied the navignore patch, also map update from mibsolution.one works.

My end goal is to build a ifs-root patch for my unit. So is there any patch-building how-to laying around?
I already uploaded my backup to mibsolution.one, if necessary I could share it here too.

I'm attaching a few pictures of my unit so people can see that I didn't the get SW Train wrong. :)

[image]
[image]
[image]

harman-f, 2023-05-16:
get a backup please

zzz08900, 2023-05-16:
> get a backup please

A backup zip along with its logs are already uploaded to https://mibsolution.one/#/1/14
Filename: MU1002-MHI2_CN_VW511_P1002-AW7-MON22.08.1800061022.zip ---> the backup zip
Filename: MU1002-AW7-MON22.08.1800061022.log ---> the log file

harman-f, 2023-05-16:
1st time I see this train.
Could someone already changed this?

zzz08900, 2023-05-16:
> 1st time I see this train. Could someone already changed this?
> 
> Which car model is it?

Well, I did a few map updates and applied navignore patch. Everything else should be factory.
The car is a 2018 Phideon, a SAIC-VW special that's (afaik) only sold in China.

harman-f, 2023-05-16:
Check M.I.B folder on mibsolution ;)

zzz08900, 2023-05-16:
> Check M.I.B folder on mibsolution ;)

Will test it tomorrow. You are really a miracle worker.
A thousand thanks.

zzz08900, 2023-05-17:
Tested and verified that the patch works. I have successfully added sport-MHI FEC to my MHI2 unit. 
The only reading I can get from sport-MHI is coolant temp - no surprise here since sport-MHI was never meant for my car model.

closed 2023-05-17 by zzz08900