Mr-MIBonk/M.I.B._More-Incredible-Bash #216: Header seize on unit does not match patch files seize

junibuNn, 2023-05-18:
Hey,
I tried to flash my unit with the patched fw i got from mibsolution "MHI2_ER_AUG22_P3241_MU0678".
Unfortunately I get this Error message...

[MU0678-JTB-07413.09.161190031C.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11511165/MU0678-JTB-07413.09.161190031C.log)

best regards

harman-f, 2023-05-18:
something is off with your patch file.

Show a screenshot of your patch folder please:
patches/MHI2_ER_AUG22_P3241_MU0678_PATCH

junibuNn, 2023-05-19:
[image]

harman-f, 2023-05-19:
Not sure what you are doing here, but that is 100% wrong. 

Check wiki section here. 
there is a guide on how to setup the SD card.

closed 2023-05-19 by harman-f