Mr-MIBonk/M.I.B._More-Incredible-Bash #217: 048 tmc or tpeg

INH888, 2023-05-19:
Bro, can the software flash TPEG? That is, to update the real-time traffic information on the 048 AISIN head unit.

harman-f, 2023-05-19:
Can you give more details on this?

INH888, 2023-05-21:
> Can you give more details on this?

[image]
Like that, it shows real-time traffic information on the navigation

catatonicChimp, 2023-05-21:
1. What country?

TPEG is a lot different to RDS-TMC in terms of implementation, there is set servers which the car/headunit vin has to authenticate against before data is returned... short answer is probably a no...

INH888, 2023-05-22:
> 1. What country?
> 
> TPEG is a lot different to RDS-TMC in terms of implementation, there is set servers which the car/headunit vin has to authenticate against before data is returned... short answer is probably a no...

CHINA.
Dude, you mean it's not free?

catatonicChimp, 2023-05-22:
Depends on the data supplier and the way its setup with VW
TMC-RDS is specifically over FM radio which makes it a lot easier to hack around since its a one way transaction and all the location data is stored locally. 

TPEG can be over either DAB+ Radio or more likely via IP (Wifi/Mobile Data) 
In the case of VW it appears all TPEG data comes via IP (Wifi/Mobile Data) and given the LTE icon in your photo above, it looks like you need a LTE connection for it to work. 

From my brief reading it requires to authenticate against a VW License server checking both VIN + License before it allows you to connect to the data provider, I believe it used to be Tomtom but that is the limit of my knowledge because VW never enabled it in my country. Even if I could hack it to work like some people tried on the Audi Forums it wouldn't work for me..  even if you get the config correct, you also need to hack a VIN into the radio that has a online entitlement already, and it not been expired, it also sounds like VW has blacklisted any unit that didn't originally come with a 3G/LTE modem integrated.

harman-f, 2023-05-22:
[image]

[image]

INH888, 2023-05-23:
> Depends on the data supplier and the way its setup with VW TMC-RDS is specifically over FM radio which makes it a lot easier to hack around since its a one way transaction and all the location data [...]

To my knowledge, someone has achieved real-time traffic information feature on a non-SIM enabled Harman 048 head unit through the WLAN function in the navigation interface.  However, they did not provide a detailed explanation on how it was accomplished.

INH888, 2023-05-23:
> [image]
> 
> [image]

Bro, I apologize for my lack of proficiency with the program.  Has the program already been updated?  And if so, how can I use it?

INH888, 2023-05-23:
> [image]
> 
> [image]

Bro, I apologize for my lack of expertise with the program.  Has the program been updated?  And if so, could you please guide me on how to use it?

harman-f, 2023-05-23:
No, just some strings I found in the existing files referencing `freeDAB-TPEG`

Get us a backup of such an improved unit ;)

closed 2023-06-08 by harman-f

qf512415007, 2024-06-09:
这个怎么更新

qf512415007, 2024-06-09:
> > Can you give more details on this?
> 
> 
> 
> [image]
> 
> Like that, it shows real-time traffic information on the navigation
> 
> 

你最后找到方法没有

isarrider, 2025-01-14:
Did this go somewhere?