Mr-MIBonk/M.I.B._More-Incredible-Bash #221: How can I replace the oem carplay icon on MIB2 high with some SD script?

Nathan2090, 2023-05-22:
Hello, is there any way to replace the oem carplay icon on MIB2 high with some SD script? I found it's file path is /etc/eso/media/carplay_oemicon.png, but I cannot replace it with any SD script tools.....

harman-f, 2023-05-22:
In General icons can be replaced. Methodists are different based on FW/Brand. 

Which train do you have. 

What do you want to replace it with?

Nathan2090, 2023-05-22:
> In General icons can be replaced. Methodists are different based on FW/Brand.
> 
> Which train do you have.
> 
> What do you want to replace it with?

My software version is MHI2_CN_VW48x_P0841_MU0841, and I want to use the new VW MIB3 CarPlay icon to replace the old one…

harman-f, 2023-05-22:
carplay_oemicon.png is th eone displayed in CP to go back to your MIBs screen.

The one from the MIB GUI is hidden in a skin file.

mib2toolbox has tools for that:
https://github.com/jilleb/mib2-toolbox/tree/master/Tools

Nathan2090, 2023-05-22:
> carplay_oemicon.png is th eone displayed in CP to go back to your MIBs screen.
> 
> The one from the MIB GUI is hidden in a skin file.
> 
> mib2toolbox has tools for that: https://github.com/jilleb/mib2-toolbox/tree/master/Tools

Yes, I know those tools and the MIB GUI can be moded by them. But the icon (VW logo) displayed in CarPlay home page is located in /etc/eso/media/carplay_oemicon.png as I found. Can I replace it with some SD scripts?

harman-f, 2023-05-22:
[image]

[image]

Create a script file called: `command.sh`
and place it in `/mod/` on SD card

You can the script from GEM
`Main -> M.I.B. -> Advanced Settings -> Individual Script`

Nathan2090, 2023-05-22:
> [image]
> 
> Create a script file called: `command.sh` and place it in `/mod/` on SD card
> 
> You can the script from GEM `Main -> M.I.B. -> Advanced Settings -> Individual Script`

Okay, I'll try this. Thanks a lot.

harman-f, 2023-05-22:
How does the MIB 3 icon look like?

Nathan2090, 2023-05-22:
> How does the MIB 3 icon look like?

[image]
like this one...

harman-f, 2023-05-22:
Ah, you want the new logo tyle ;)

[image]

Nathan2090, 2023-05-22:
> Ah, you want the new logo tyle ;)
> 
> [image]
> 
> In case you need the old one: [carplay_oemicon.zip](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11527798/carplay_oemicon.zip)

thanks :)

closed 2023-05-27 by Nathan2090