Mr-MIBonk/M.I.B._More-Incredible-Bash #223: After modified the shadow.txt to replace the root password, the unit reboot every 60s....

Nathan2090, 2023-05-22:
Hello~~After modified the shadow.txt to replace the root password, the unit reboot every 60s. The unit can enter the system menu and can be operated, but automatically reboots every 60s....How can I fix it?

harman-f, 2023-05-22:
With M.I.B function?

Please share LOG file

Nathan2090, 2023-05-22:
> With M.I.B function?
> 
> Please share LOG file

actually, I patched it with mib2-toolbox....

zzz08900, 2023-05-22:
Looks like some kind of system integrity protection stuff. Try place the original file back maybe?

harman-f, 2023-05-22:
That could sadly be the case. 
was the unit patched before?

please Share your backup from SD.

zzz08900, 2023-05-22:
Just asking, does these systems has a flash chip that can be re-programmed by an external chip programmer?
Or is the unit doomed if we mess up something in a rather bad way?

harman-f, 2023-05-22:
There are levels of bad 😉 
Many can be recovered.

zzz08900, 2023-05-22:
OK I got it. There's no "last resort one key reset all" around.

Nathan2090, 2023-05-23:
> That could sadly be the case. was the unit patched before?
> 
> please Share your backup from SD.

It's just an original shadow file....
[shadow.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11544280/shadow.txt)

zzz08900, 2023-05-24:
You could try the on-board telnet connectors but they may not work. My guess is telnet related stuff is what triggers integrity protection.

Nathan2090, 2023-05-24:
> You could try the on-board telnet connectors but they may not work. My guess is telnet related stuff is what triggers integrity protection.

May be you are right. The root password is protected by some security mechanism on my unit.

closed 2023-05-27 by Nathan2090