Mr-MIBonk/M.I.B._More-Incredible-Bash #237: Patch needed for MHI2_US_VWG13_P3662_1 (MU1418)

mc32768, 2023-06-05:
I see there is already a patch for MHI2_US_VWG13_P3662, but I have MHI2_US_VWG13_P3662_1 (same thing but with "_1" at the end).  The patch for MHI2_US_VWG13_P3662 does not seem to work on my "P3662_1" device.

I have placed a backup here and attached logs and screen images:
https://filehorst.de/d/eHumgmvw

Is a new patch required for the "_1"?

Thank you for your help!

[image]

[image]

[MU1418-JTB-07419.09.19119403C0.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11657796/MU1418-JTB-07419.09.19119403C0.log)

harman-f, 2023-06-05:
You can just rename the existing patch folder.

mc32768, 2023-06-06:
That worked - thank you so much!  This software is amazing and I am very grateful to have access to it.

closed 2023-06-06 by mc32768