Mr-MIBonk/M.I.B._More-Incredible-Bash #246: Cannot switch on User Defined in Red Menu

AudiFun, 2023-06-11:
I'm using MIB 3.5.1 and during conversion from MU1320-MHI2Q_US_AUG22_P4246 to MHI2Q_ER_AUG22_P5092_MU1329 I cannot switch on SWDL User Define mode. In red menu this option is always gray, I've only "Standard" option available.
I have to underline that I'm switch on in MIB opitions "User Defined" nad "Ignore Variant and Region".
Til now I have done:
- Automati instalation of MIB 3.5.1
- Installation of EU Train from MIB

Third step as I understood should be installation EU firmware and it is not possible.

AudiFun, 2023-06-11:
I do not know if it helps but during automatic installation of MIB, in patch folder I had already PATCH of my new FW (MHI2Q_ER_AUG22_P5092_MU1329).
To be honest I do not know what MIB are doing during automatic installation but maybe this caused my problem.

harman-f, 2023-06-11:
I do not see any conversion attempts in your LOG file.

Best would be to switch to latest M.I.B directly from repo.
EU conversion changed a lot in the last week.

https://mibwiki.one/doc/conversion-of-cnjpkrtwus-to-eu-8lv9Zw5DzP

AudiFun, 2023-06-12:
Sorry, I didn't copy my complete LOG file. I have only available the below one.
Even in this maybe there isn't conversion. Probably I lost this file when I've tried with new MIB on SD.

Now, when I'm doing again conversion I have feedback that EU train already exist and stop the process. 
Can I copy your recomended M.I.B from mibsolution.one ?

Logging_MU1320-MHI2Q_US_AUG22_P4246-YJL-26001.06.186969021F:
HW Number: 034 - PN: 8W5035880D
Variant: FMQ-P-TNSL-US-AU-MLE
EEPROM Variant: MU-P-LNS-US
Dataset Version: 80B909545N
Model ID: 012406 - Car Model: Audi Q5 (8W5)
VIN -> CAR:  - MIB: 

 ls --> total 2303
[image]

[image]

[image]

       8 drwxrwxrwx 11 root      root           4096 Jan 01  1970 .
       8 drwxrwxrwx 11 root      root           4096 Jan 01  1970 ..
       8 drwxrwxrwx  2 root      root           4096 May 31 20:32 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Apr 30 19:33 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Apr 30 19:33 .gitattributes
       8 drwxrwxrwx  3 root      root           4096 Apr 30 19:33 .github
       1 -rwxrwxrwx  1 root      root            132 Apr 30 19:33 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Apr 30 19:33 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Apr 30 19:33 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Apr 30 19:33 LICENSE
       8 drwxrwxrwx  4 root      root           4096 Apr 30 19:33 Launcher
       1 -rwxrwxrwx  1 root      root             60 Apr 30 19:33 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Apr 30 19:33 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Apr 30 19:33 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6438 Apr 30 19:33 README.md
       0 -rwxrwxrwx  1 root      root              0 Apr 30 19:33 Swdlautorun.txt
       1 -rwxrwxrwx  1 root      root             24 Apr 30 19:33 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
       8 drwxrwxrwx  3 root      root           4096 Apr 30 19:33 apps
       8 drwxrwxrwx  4 root      root           4096 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Apr 30 19:33 bash.png
       8 drwxrwxrwx  2 root      root           4096 Apr 30 19:33 config
       8 drwxrwxrwx  3 root      root           4096 Apr 30 19:33 esd
       4 -rwxrwxrwx  1 root      root           1995 Apr 30 19:33 metainfo2.txt
       8 drwxrwxrwx 10 root      root           4096 Apr 30 19:33 mod
       8 drwxrwxrwx  4 root      root           4096 Apr 30 19:33 patches
      16 -rwxrwxrwx  1 root      root           7874 Apr 30 19:33 start

RCC-backup---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1acdf6b (15916908)|93...215 (122):LOCKED
0x00ba0000 - 6CDFF200

Found YJL-26001.06.186969021F Backup, will check if all complete...
Version info..... -> done :-)
FEC folders...... -> done :-)
Shadow files..... -> done :-)
Dataset's........ -> done :-)
EEPROM........... -> done :-)
storage1......... -> done :-)
storage2......... -> done :-)
backing up backup -> part 1 done
ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  4 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1320
Free space on /net/mmx/mnt/boardbook 992M

RCC-launcher---->
[... 75 lines cut ...]

ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  4 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU1320
Free space on /net/mmx/mnt/boardbook 928M

Reboot Unit...

RCC-svm---->
Fix SVM Error (Note: The process can take some minutes...)
extract build version...
SvmInfo - Compressing... /net/rcc/mnt/efs-persist/SWDL/Log/SvmInfo.bin CfgAckRand - Uncompressing...
 /net/rcc/mnt/efs-persist/SWDL/Log/CfgAckRand.z to /net/rcc/mnt/efs-persist/SWDL/Log/CfgAckRand.bin

Key: 4CF3 - CHKSum: 8521

CfgAck - Compressing... /net/rcc/mnt/efs-persist/SWDL/Log/CfgAck.bin Wait for DSI to connect ...
DSI Connected role1.DSIE2PAccess
responseWriteE2P address 0x3f0 status 0 len 0x03
0x3f0 	00 00 01 
DSI Disconnected
clear SVM done!

Reboot Unit...

AudiFun, 2023-06-12:
I did again EU conversion from M.I.B 3.6.0 and problem doesn't solved. 
I cannot instal EU FW.
[image]

harman-f, 2023-06-13:
get latest version directly from repo: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/archive/refs/heads/main.zip

3.6.0 might have a bug

R3DcooDe, 2023-06-13:
> I did again EU conversion from M.I.B 3.6.0 and problem doesn't solved. I cannot instal EU FW. [image]

bro do not complicate things - after eu convert - you have to enable skip region check and user defined / after press radio button  do not restart unit   just pres radio -  remove MIB sd card from car insert SD with firmware - go to red menu and update

harman-f, 2023-06-13:
Looking at his log only the 1st script is running from the EU conversion.
This seems to be a bug fixed with latest version from repo.

AudiFun, 2023-06-13:
> 

Believe me, I did as you write, User defined is always gray in red menu. I do not know what's going on ?!

AudiFun, 2023-06-13:
@harman-f 
Today, again EU conversion (M.I.B from repo) and User defined in red menu still gray.

Below my today's LOG :

 __  __     ___           ___  
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - V3.6.X "TEST Edition"
______________________________________________________________________

NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF

Logging_MU1320-MHI2Q_ER_AUG22_P5092-YJL-26001.06.186969021F:
PN: 8W5035880D - HW Number: 034
Metainfo2 Variant (Variant2): FMQ-P-TNL-EU-AU-MLE
EEPROM Variant (Variant1): MU-P-LN-EU
Parameter set part number: 80B909545N
Coding: 012406010000000013110002041a00102f0015e501700904be
Car Model: Audi Q5 (8W5)

______________________________________________________________________

ls -->
total 2567
      32 drwxrwxrwx 11 root      root          16384 Jan 01 00:00 .
      32 drwxrwxrwx 11 root      root          16384 Jan 01 00:00 ..
      32 drwxrwxrwx  2 root      root          16384 Jun 13  2023 .System Volume Information
       2 -rwxrwxrwx  1 root      root            986 Jun 13  2023 .git-blame-ignore-revs
       1 -rwxrwxrwx  1 root      root            399 Jun 13  2023 .gitattributes
      32 drwxrwxrwx  3 root      root          16384 Jun 13  2023 .github
       1 -rwxrwxrwx  1 root      root            132 Jun 13  2023 .gitignore
    1703 -rwxrwxrwx  1 root      root         871810 Jun 13  2023 2022_06_MHI2_Password_List_V4.0.pdf
      81 -rwxrwxrwx  1 root      root          41255 Jun 13  2023 GEM.png
      30 -rwxrwxrwx  1 root      root          14939 Jun 13  2023 LICENSE
      32 drwxrwxrwx  4 root      root          16384 Jun 13  2023 Launcher
       1 -rwxrwxrwx  1 root      root             60 Jun 13  2023 M.I.B. - More Incredible Bash - Mibsolution Wiki.url
       0 -rwxrwxrwx  1 root      root              0 Jun 13  2023 NOT FOR COMMERCIAL USE - IF YOU BOUGHT THIS YOU GOT RIPPED OFF.txt
     284 -rwxrwxrwx  1 root      root         144962 Jun 13  2023 PATCH COMPATIBILITY TABLE.pdf
      13 -rwxrwxrwx  1 root      root           6438 Jun 13  2023 README.md
       1 -rwxrwxrwx  1 root      root             22 Jun 13  2023 VERSION
       1 -rwxrwxrwx  1 root      root            296 Jan 01  1980 WMPInfo.xml
       0 -rwxrwxrwx  1 root      root              0 Jun 13  2023 _Swdlautorun.txt
      32 drwxrwxrwx  3 root      root          16384 Jun 13  2023 apps
      32 drwxrwxrwx  3 root      root          16384 Jan 01  1980 backup
      77 -rwxrwxrwx  1 root      root          39351 Jun 13  2023 bash.png
      32 drwxrwxrwx  2 root      root          16384 Jun 13  2023 config
      32 drwxrwxrwx  3 root      root          16384 Jun 13  2023 esd
       4 -rwxrwxrwx  1 root      root           2040 Jun 13  2023 metainfo2.txt
      32 drwxrwxrwx 11 root      root          16384 Jun 13  2023 mod
      32 drwxrwxrwx  3 root      root          16384 Jun 13  2023 patches
      16 -rwxrwxrwx  1 root      root           7874 Jun 13  2023 start

______________________________________________________________________

timestamp: 1970_01_01_00_02_48----->
MMX-settrain---->
Current train: MHI2Q_ER_AUG22_P5092

Skipping. Your train is already EU!

timestamp: 1970_01_01_00_02_51----->
MMX-setreg---->
Writing EU language datasets to 5F...

timestamp: 1970_01_01_00_02_54----->
MMX-setlang---->
Writing EU dataset 0x2D00 (visible_MMI_language_dataset):
de_DE,en_GB,en_SA,fr_FR,it_IT,es_ES,pt_PT,pt_BR,nl_NL,ru_RU,pl_PL,
cs_CZ,es_MX,en_US,sv_SE,tr_TR,hu_HU,no_NO,ar_SA,da_DK,ms_MY,ro_RO,
sl_SI,fi_FI,el_GR,uk_UA
Writing en_GB dataset 0x280 (language dataset)

setlang done!

Changing coding and adaptations to EU:
--- Summer time algo
--- FM Band setting
--- AM Band setting
--- RDS switch off
--- AF functionality on
--- Radiotext+ on
--- PI Ignore off
--- AM disable on
--- RDS deactivation off
--- AF deactivation off
--- Navigation Country
--- Import media
--- Media Ripping
--- Regional Codes DVD
--- Regional Codes Blueray
--- Regional Codes DVD Counter
--- Regional Codes Blueray Counter
--- TMCPro EU
--- Gracenote local ON
--- Radio stationlogo DB
--- SDS region Flag
--- Variant region

setreg done!

timestamp: 1970_01_01_00_03_01----->
MMX-setvariant---->
Сurrent Variant_1: MU-P-LN-EU
Current Variant_2: FMQ-P-TNL-EU-AU-MLE
Reading Variant_2 from EEPROM...
Reading Unit Class from EEPROM...
Reading Feature bits from EEPROM...
Reading Region from EEPROM...
Cleaning Variant_1 in EEPROM(writing 13 zero bytes)
Wait for DSI to connect ...
DSI Connected role1.DSIE2PAccess
responseWriteE2P address 0xba status 0 len 0x0d
0xba 	00 00 00 00 00 00 00 00 00 00 00 00 00 
DSI Disconnected
Variant_1 generated from Variant_2: MU-P-LN-EU
Writing generated Variant_1 into EEPROM
Wait for DSI to connect ...
DSI Connected role1.DSIE2PAccess
responseWriteE2P address 0xba status 0 len 0x0a
0xba 	4d 55 2d 50 2d 4c 4e 2d 45 55 
DSI Disconnected
New Variant_1 in EEPROM: MU-P-LN-EU

setvariant done!

Now you can insert FAT32 formatted SD card 
with FW MHI2Q_ER_AUG22_P5092 and update.
IMPORTANT - If available use AIO FW version

harman-f, 2023-06-13:
Script looks much better now :)

So you go into GEM - M.I.B menu and enable skip region check and user defined.
Exit GEM (no reboot)
And enter REM to update FW?

AudiFun, 2023-06-13:
> 

Exactly, this I did. The same slot for M.I.B and FW - SD1

harman-f, 2023-06-13:
No idea in that case.

To edit metainfo2 is the other option.

AudiFun, 2023-06-14:
I don't want to brick my MMI, please advice me with which option I should start from.
Now I can try anything.

R3DcooDe, 2023-06-14:
can you update via standart , if you need user defined mode to avoid bose update try using a custom update , bose is not selected by default , i think

harman-f, 2023-06-14:
Doing standard can end in a mess.
Only FW components with higher version will be installed.

And the current FW might have some newer components than the new target FW.

Post your metainfo here, I will try to find some time tomorrow to edit it.

AudiFun, 2023-06-15:
After post of @R3DcooDe I did Standard :)
I don't have radio, navi, and I don't know what else.

Unfortunately I do not know how I can get metainfo from my MMI

R3DcooDe, 2023-06-15:
damm bro im really sorry for your loss , i messed up - i forgot your are trying to convert unit, not just update from europen old version to Europeans new version..  i hope @harman-f  can find you a solution to bring mmi back to life via uart .  im really sorry - i wont give anymore advices in future :/

harman-f, 2023-06-15:
> After post of @R3DcooDe I did Standard :( I don't have radio, navi, and I don't know what else. Also there is no 5F visible now in VCDS/VCP/OBD11. In the previous car I've unlocked my MMI by USB=>UART method. I do not see any other option now,
> 
> @harman-f Unfortunately I do not know how I can get metainfo from my MMI

Can you still get into update menu?
Or is unit locked on on escreen and rebooting like every 60s?

Describe the current state in a bit more detail.

If you have UART/D-Link log in to unit and get us a copy of `/net/rcc/mnt/efs-persist/SWDL/update.txt`.

AudiFun, 2023-06-17:
> Can you still get into update menu? Or is unit locked on on escreen and rebooting like every 60s?
>
> Describe the current state in a bit more detail.
>
> If you have UART/D-Link log into unit and get us [...]

I did another thing:
- automatic installation of M.I.B (Swdlautorun)
- after installation enable skip region check and user defined in M.I.B
- in REM User defined was available !!!!
- marked all items without Bose and start update EU FW
- installation finish with success
- PATCH
- Map installation

It seems that everything is OK, only I do not have AM in Radio, but it's not important for me , I do not use it.

Thank you for support !

AudiFun, 2023-06-17:
All my mess with conversion probably caused by VIM installation on US software.
I do not why but during VIM enable M.I.B has checked if there is EU Train and did something.

You can check if I'm right :)

closed 2023-06-17 by AudiFun

harman-f, 2023-06-17:
VIM does not check for EU ;)
Must have been soemthing else.

Did you ever share your backup?
Make a new one and I can check if coding/variant is messed up somewhere.

AudiFun, 2023-06-18:
Understood, Here is my original backup:
https://mibsolution.one/#/1/14/New%20Folder1%20M.I.B_Backup

AudiFun, 2023-06-20:
> MMX-vim----> VIM on Unit: 01000F0AFF00000000000100FF00010001000F050F05FF0000053234AD15 Position->0 - HEX-Value->c7
>
> Generate crc16/CCITT-False checksum............................... and the Winner [...]

Please look at my LOG - here is the info about checking EU by VIM.

harman-f, 2023-06-20:
you started multiple scripts at the same time.

The EU message is from EU conversion script...