Mr-MIBonk/M.I.B._More-Incredible-Bash #263: Patch not working on multiple firmwares

lockdown-ng, 2023-06-27:
So here is the thing..
I had an original firmware MHI2_ER_AUG11_P0034 on my Audi A7..
I then downgraded to "MHI2_ER_AU57x_K2589" as it was instructed in the Readme of AIO patched firmware  MHI2_ER_AU57x_K3663_1_MU1425_AIO

Finally, I installed MHI2_ER_AU57x_K3663_1_MU1425_AIO. 
I noticed my FEC file is not changed and there is no AndoridAuto or CarPlay activated. I tried with M.I.B but got failed messages as well.
I enabled AndroidAuto through VCDS so now I can just see it, but it can't see any phones attached, and FEC file is the same as it was before any upgrades....
The system shows the new firmware version after every update, but everything remains the same :/
I am also surprised why the HI2_ER_AU57x_K3663_1_MU1425_AIO firmware didn't work  to patch my unit, and what can I do to make it work once the MIB supports my unit?

I tried MIB on all of the mentioned Firmwares, and it said it doesn't support either of them.
I also tried this firmware:  MU0335-MHI2_ER_AUG11_P0040 for which I also downloaded the full M.I.B. backup....
So would it be possible to tell me which files should I upload here in order to see where the problem is?

I am attaching few logfiles...If anything else is needed, please let me know..

[MU0335-JTB-07411.03.1511710205.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11876805/MU0335-JTB-07411.03.1511710205.log)
[MU0335-version.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11876821/MU0335-version.txt)
[MU0335-allversions.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/11876822/MU0335-allversions.txt)

harman-f, 2023-06-27:
Your train is not changed and many other things. 

Install the 3663 AIO again.

lockdown-ng, 2023-06-27:
What will reinstalling AIO do when the built in patch doesn't work? 
And MIB doesn't work on that firmware either... Maybe because it is AIO patched firmware version, or because of some other reason.. Don't know... I tried already...
I can try again to update it from User mode and post MIB logs after it fails... Just let me know which files to upload...

harman-f, 2023-06-27:
Normally, the AIO does not fail. 
Run again, that should work.

lockdown-ng, 2023-06-28:
well, I tried again but failed... I then ran MIB so it would make some logs... They are uploaded here:
https://mibsolution.one/index.php/f/182244
I will also upload some pictures soon..

harman-f, 2023-06-28:
Check patch folder on SD
You have to add the "_1" to folder to match train name.
[image]

It is strange, that patching and adding FEC with AIO is not working.
Something is off here

closed 2023-06-29 by harman-f

lockdown-ng, 2023-06-30:
Where is this patch folder? All I see on SD(with MIB on it) is this, and it is empty...

[image]

harman-f, 2023-06-30:
The _1 has to go behind the …K3663_1

for now pre made patch files are still needed. 
but hopefully not for long anymore.

lockdown-ng, 2023-06-30:
OK, I had to actually name the folder `...K3663_1_MU1425...` instead of `...K3663_MU1425_1....`
And the MIB finally showed me that my firmware has already been patched, as it should have been since I used `3663 AIO Custom Update`... If you are also an author of this Firmware patch: `https://mibsolution.one/au_3663_aio`, I suggest to check if some Sleep() or something needs to be added because the patch apparently worked, but FECs didn't(and other things, don't know how to check).

So, I just tried to manually patch FECs from MIB and it seemed it worked as can be seen on the pictures below... 
The car play started to recognize phones now as well...

But it still seems like there are 2 more things to solve in case you can help out:
1. my maps still won't update, as shown on the picture as well.... I downloaded `EU full` maps from here:
http://mib-helper.com/index.php?train=MHI2_ER_AU57X_K3663
2. Gracenote2  - I don't know how to check if it's installed on not? Where should the icon be or what should I see?

**PATCH:**
[image]

**FECs:**
[image]
[image]

**Navigation:**
[image]

[image]

lockdown-ng, 2023-06-30:
all done... Either I had to activate something through MIB2 Toolbox or I had to put MIB1 and MIB2 folder onto SD card, which I previously didn't(I've put only MIB2 folder from the NAV update ZIP file)...
So everything is working now...
Thanks for the help...

harman-f, 2023-06-30:
you need both MIB1 and MIB2 folder ;)

That explains a lot!