Mr-MIBonk/M.I.B._More-Incredible-Bash #265: Mhi2_us_aug11_s0533 - Not Supported

R3DcooDe, 2023-06-28:
Hello Sir 

Today i was trying to convert AUDI A6 2016  from USA
with firmware Mhi2_us_aug11_s0533 to EUROPE and install European firmware : MHI2_ER_AU57x k3663 because i couldn't find any update starting with mhi2_er_aug11xxxxx 

I installed mib from sd card to car.  I opened mib i clicked change train to eu.  All stets said done. Until last step said something not found. But it move to fast and restarted i couldn't see what exactly was.      So i run mib again and i clicked 5minute backup so , i will provide back up to you and logs from mib sd card.   I hope you can help me with patch for this unit to successfully convert to europe and install eu maps.  

Thank you 
Backup link here : 

https://www.mediafire.com/file/nx17cfsnkhi3nbm/backup+mib.zip/file

[image]

harman-f, 2023-06-28:
AU57x is correct for aug11

Car is in EU now?

Did you do a conversion before installing the EU FW?

R3DcooDe, 2023-06-29:
@harman-f  yes i installed MIB and clicked conversion to europe. But mib fail to do it last step it says ,  there is no conversion train found PLEASE CONTACT MIB creators for Help.      So car it is with original firmware now.   Please see logs and backup on the link above . 

Thank you

harman-f, 2023-06-29:
Conversion is not yet merged with main. 
will need a few more days. Currently focusing on flash.

R3DcooDe, 2023-06-29:
@harman-f  okay sir , so i have to wait a few days. And you will add  this firmware to supported list

harman-f, 2023-06-29:
please create a fresh M.I.B backup and share (filehorst.de)

I need to check which state the unit is in ;)

R3DcooDe, 2023-06-30:
@harman-f  here is the link : https://filehorst.de/d/erzekfgi

harman-f, 2023-06-30:
OK, one of the issues is, that you are using a quite old M.I.B version.
It is not supporting your very old FW.

Get latest version from repo directl, it will support it.

Before converting to EU you should install latest US FW.
M.I.B will fully work here.

To be able to do this you have to edit metainfo2 to enable install over AUG11 train and disable BOSE.

R3DcooDe, 2023-06-30:
hello mate thank you for your time, i actually tried today this method , i install latest us firmware MHI2_US_AU57x_K3341  and installed mib again, run same steps like always ,  first eu conversation , after step 1, 2 eeprom and 3 reboot,   deleted nav database.  /   now the fun part is when i open red menu to install European firmware .  it says  MHI2_EU_AU57x_K3663 / MU Software 1426 ---- -  on the other hand European firmware on sd card it has MU Software 1425 

now when i go to red menu to update there is no USER DEFINED MODE  its greyed out..  i tick skip region check and enable user defined  at mib advance swdl ..  

now im stuck here cant install european firmware 

[image]

closed 2023-07-08 by R3DcooDe