Mr-MIBonk/M.I.B._More-Incredible-Bash #271: after updating patch I have lost the juke box juke functionality as well as recording; also dvd movies files are not readable.

Alussu, 2023-07-03:
I had the mib2 high working properly, I did a restore and I have lost the juke box menu, no function to copy from usb memory or CD/DVD. I can see the menu to check the juke box memory which seems OK (it is empty because I formatted it). 
In GEM the media copying functions are enabled. 
My unit was mhi2q_er_aug22_p5092_mu1329
I patched it with the same file. 
I have restored the configuration from my backup. 
This problem happened before, but I managed to get it back by going through my 5F obdeleven full backup restoring line by line the configuration oh the long coding and adaptations. this time it doesn't work and I need a help. 
Thanks

harman-f, 2023-07-09:
patch is just applying the patched ifs-root-stage2 and enebaled coding for AA/CP.

No idea how this could effect what you observed.

Alussu, 2023-07-10:
Thanks for your prompt answer. 
I'm afraid it has to do in fact with something I have done in obdeleven rather than GEM. I did restore the backup within the mib2 done at first GEM installation and went through the printout of obdeleven backup and I can't go back to the Dvd movie and Juke box.
I will try something else.

Alussu, 2023-07-10:
I did use all 3 restore options but with the same result

closed 2023-07-10 by Alussu