Mr-MIBonk/M.I.B._More-Incredible-Bash #282: MU0918-MHI2Q_ER_AUG22_P3638 not patchable

Galvanize96, 2023-07-14:
Hi,

first of all, thank you for your hard work and your great tool.

I was trying to update the MIB Systen.
My car is a imported canadian spec car converted to EU. The conversion was done before i got the car. 
Thats why i think there is a weird Version currently installed.
The current MIB version is MU0918-MHI2Q_ER_AUG22_P3638. When trying to patch the MIB, i am getting a error stating, that there was no patch available.
All available patches were in the patches folder.

Please see the log file 
[MU0918-JTB-07420.03.171171022C.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12052929/MU0918-JTB-07420.03.171171022C.log)

Is there a way to still update the MIB?

Maybe this is also helpfull:
[MU0918-allversions.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12052970/MU0918-allversions.txt)

Thank you for your help in advance.

harman-f, 2023-07-14:
Quite old FW...
That is the reson no patch was made so far.

You could update FW and patch that.

Galvanize96, 2023-07-14:
> Quite old FW... That is the reson no patch was made so far.
> 
> You could update FW and patch that.

Is that something, that the MIB tool is capable of doing?
Or do i have to get some HW to update the FW?

If it is capable of doing it, is there a instruction somewhere?

Galvanize96, 2023-07-15:
So i just wanted to inform you, that i updated the FW to MHI2Q_ER_AUG22_P5092_MU1329 and installed everything.

Thank you again for this great tool.

closed 2023-07-15 by Galvanize96