Mr-MIBonk/M.I.B._More-Incredible-Bash #283: M.i.b will not appear in the green menu

Vesawor, 2023-07-17:
Hi, i did everything according to the instructions, i installed m.i.b on the sd card. The update was successful, I activated the developer menu and after I open it, m.i.b is not there. Tried another card, reinstalled and still nothing.

Gem 4.5st if you need to know
MHI2_ER_VWG11_K3342_MU1427

Thank you so much.

harman-f, 2023-07-17:
Did you leave SD card in unit?
No write protection on SD card?

Vesawor, 2023-07-17:
Yes, i leave it in. 
No write protection.

harman-f, 2023-07-18:
Get the log file and share here

Vesawor, 2023-07-20:
Sorry, i dont know how to do it.

lprot, 2023-07-21:
> Sorry, i dont know how to do it.

zip and attach backup folder from SD. If there is no backup folder that means SD was write protected (slider on the left was in down position) when you inserted it into SD slot.

closed 2023-07-23 by harman-f