Mr-MIBonk/M.I.B._More-Incredible-Bash #290: Modify datasets in MIB unit - possible?

zzz08900, 2023-07-26:
What I'm trying to achieve:
I added central speaker and subwoofer to my car and I'm trying to flash a modified 0x003000 dataset into my unit.

I tried the "Run individual script" function with the following script
`export PATH=:/proc/boot:/sbin:/bin:/usr/bin:/usr/sbin:/net/mmx/bin:/net/mmx/usr/bin:/net/mmx/usr/sbin:/net/mmx/sbin:/net/mmx/mnt/app/armle/bin:/net/mmx/mnt/app/armle/sbin:/net/mmx/mnt/app/armle/usr/bin:/net/mmx/mnt/app/armle/usr/sbin`
`export LD_LIBRARY_PATH=/net/mmx/mnt/app/root/lib-target:/net/mmx/mnt/eso/lib:/net/mmx/eso/lib:/net/mmx/mnt/app/usr/lib:/net/mmx/mnt/app/armle/lib:/net/mmx/mnt/app/armle/lib/dll:/net/mmx/mnt/app/armle/usr/lib`
`export IPL_CONFIG_DIR=/etc/eso/production`

`#write dataset`
`on -f mmx /eso/bin/apps/pc b:0:0xC0030014 {dataset_content_in_hex}`
`#store changes to persistence`
`on -f mmx /eso/bin/apps/pc b:0:1 0`

`#reboot to save settings to unit!`
`on -f rcc /usr/apps/mib2_ioc_flash reboot`

I saw a bunch of error message on screen and it looked like nothing was changed after reboot.

The above script came from here
https://github.com/NumberOneBot/mqb-mib-sound-datasets/blob/main/5F_MIB2-datasets.txt

Any idea what I did wrong?

zzz08900, 2023-07-26:
I noticed there's a newly added function which adds subwoofer gain to existing dataset and that is very similar to what I'm trying to do. That script calls $PC to write the dataset but I couldn't find what that $PC variable is set to.

What am I missing here?

Muilperen, 2023-07-26:
yes, it is possible.
check this one first=> https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/discussions/205#discussioncomment-6074238
My mistake was to use a windows text editor. That won't work as the script should be saved in the unix EOL format. Not sure if that is your issue.

zzz08900, 2023-07-26:
> My mistake was to use a windows text editor. That won't work as the script should be saved in the unix EOL format. Not sure if that is your issue.

Well I use Linux-based machines as my daily drive, but when I couldn't get the script wo work I actually went out of my way and dug out a dusty windows installation. I guess things went sideways even faster from that point.

I just realize we need to use the PC program came with MIB to write the datasets and for me the path passed to on -f was wrong as ****.

zzz08900, 2023-07-26:
@Muilperen can you confirm you got your datasets written using on -f /eso/bin/apps/pc? 
I thought I tries that already and it didn't work and subwoofer script came with MIB was using something different to write dataset.

zzz08900, 2023-07-27:
OK guys I made it work, finally.
@Muilperen a thousand thanks, without your hints I would be blowing my head trying find what went wrong.

What I was referring to was if you were able to wrote dataset without changing the `on -f mmx /eso/bin/apps/pc` part prepending your dataset address and content in the script posted by NumberOneBot. It turned out we don't need to change that part.

I re-wrote my original script using my Linux daily machine (like, word to word re-type but not copy-pasting anything so I won't bring in windows EOL by accident) and I was able to write to any 5F dataset I wanted to and the only thing I need to change was address and dataset content. 

Now I have 9.1 channel output activated and voice message sound processing properly updated. Gonna need some personal sound tuning but it sounds pretty good already

closed 2023-07-27 by zzz08900