Mr-MIBonk/M.I.B._More-Incredible-Bash #291: App Connect Load time

Ed1jz, 2023-07-26:
Converted from JP to ER FW K3342 patched with MIB 3.5.1 but every time I load App Connect or Service Mode it takes time to load (approximately 1-2 minutes). Everything works fine just the load time. Can this be fixed? 

[MU1427-AW7-MON08.01.1600060958.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12172739/MU1427-AW7
[image]
[image]
-MON08.01.1600060958.log)

zzz08900, 2023-07-26:
It happens to me every time after a cold reboot (i.e. physically unplug the power cable than plug it back in) from my unit was 100% factory. I figure you almost never see what the unit does after  a cold reboot.

zzz08900, 2023-07-26:
> 

My unit always load the module it was shutdown with first and everything else later.
Like this, if I turn off the engine, left music on and leave the car, next time the unit will load music module first when the car turns on again. If I switch to other module like map right after turning on the car I am usually greeted with the hint "Navigation is starting".

Ed1jz, 2023-07-26:
I see your point. 

The patch and FW update was successful. Like I followed all the instructions. Issue is I've converted 3 MHI2 MIB2 units and all face the same problem of App connect loading longer than expected. 

Should I post my backup here?

zzz08900, 2023-07-26:
https://mibsolution.one/
There's a M.I.B_backup folder used for storing shared backups so devs of this project can analyze them.

Ed1jz, 2023-07-26:
@zzz08900 I will upload the back up.

@harman-f yes the 3 units are stock with the same FW MHI2_ER_VWG11_K3342 converted from MHI2_JP_VWG11_S0030
I managed to update FW to k3342 then patched with MIB 3.5.1. Everything works just that cold boot which takes a whooping 2 mins to load.

harman-f, 2023-07-26:
How did you update?

what did you do to be able to install FW?

Ed1jz, 2023-07-26:
I used the AIO K3342_1 with a modified metainfo then after inquiring I reinstalled the K3342 with a modified metainfo through SWDL cause it refused to install at first.

harman-f, 2023-07-26:
Have not seen that AIO yet. 
Share the metainfo2.txt

Ed1jz, 2023-07-26:
[metainfo2.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12173691/metainfo2.txt)

harman-f, 2023-07-26:
Well a bit overkill with the options, but it should work. 

No idea….

Ed1jz, 2023-07-26:
[metainfo2.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12173815/metainfo2.txt)

I reinstalled K3342 with this metainfo2. So in short a cold boot is normal?

closed 2023-08-06 by harman-f

pixeleyesltd, 2024-10-07:
@Ed1jz @nm-correia Was this ever resolved for you? I installed MHI2_ER_VWG11_K3342 coming from MHI2_JP_VWG11_S0021 and CarPlay works but only after 5 minutes of App Connect loading.

nm-correia, 2024-10-07:
I believe that I had the same delay during the first boot. But I had no delays after that.

Ed1jz, 2024-10-07:
@pixeleyesltd also you have to install an EU map database for it to work after applying EU coding and change variant 1 in eeprom. What I did is I installed Great Britain map only and the load time reduced significantly to 10 secs.