Mr-MIBonk/M.I.B._More-Incredible-Bash #294: Unable to see the MIB menu in Green Engineering Menu

gkelly147, 2023-07-26:
**Describe the bug**
Unable to see the menu in Green Engineering Menu (Nav UP and Media UP 5 secs). 

Using ExFAT formatted 64GB SD card to update the following modules using Red Engineering Menu (Nav UP and Media DOWN 5 secs):

- 5F: MHI2_ER_AU43x_P5098_MU1339 (TT).7z (source: [mibsolution.one](http://mibsolution.one/))
- 17: 8S0 906 961 AE - VC 0296 (source: https://drive.google.com/file/d/1no6y6ZxGmQ7smsMFQzIMAfp6dEMNwzhJ/view)

Updates completed as expected through here > https://www.audiworld.com/forums/tt-mk3-discussion-207/my2016-us-tt-carplay-activation-diy-3051673/

17 update I cannot confirm, but I extracted the files from the linked VC 0296 zip file onto the SD card and completed the update process through the Red Engineering Menu. This was the only way the SD card content would be visible. 

**Check known issues before reporting**
Make sure you have checked here:
https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/wiki/Prepare-M.I.B-SD-card-&-install-M.I.B#%EF%B8%8F-trouble-shooting-%EF%B8%8F

**Add M.I.B LOG file to your report**
Always attach your M.I.B LOG file!
[backup.zip](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12177261/backup.zip)

**Screenshots**

5F Version before updates > 
[image]

5F update confirmed here > [image]

17 Update in progress > [image]

Cannot see MIB option > 
[image]

**Additional context**
Add any other context about the problem here.
Was the unit worked on before?
The previous 5F Version was MHI2_ER_AU43x_P3226 (see screenshot above) but never worked on to my knowledge. 

Is the unit stock in car or a retrofit?
Unit is stock Oct 2016 Audi TT Mk3 (8S)

gkelly147, 2023-07-26:
One more point, for the MIB Launcher Installer parameter '_Swdlautorun.txt' to 'Swdlautorun.txt'

harman-f, 2023-07-27:
M.I.B SD is also exFAT?
SD is still in unit?

closed 2023-08-06 by harman-f