Mr-MIBonk/M.I.B._More-Incredible-Bash #297: OFFTOPIC - mibwiki.one

dawidszyszka, 2023-07-28:
Hi, i have problem with login to mibwiki.one. The website showing me error i header "?notice=authentication-required". Tried witch telegram and discord.

zzz08900, 2023-07-28:
Quick guess, could be your browser or browser plugin or something else that's interfering with discord discord third party login.
Try a clean Chrome, Firefox or Edge?
Try to use your phone's mobile hotspot and see if anything changes?

dawidszyszka, 2023-07-29:
Tried with multiple browsers (also on my friends pc) , vpn and hostpot. Always the same error, it is showing only in url. I can see warning icon but without any content.
[image]
Telegram login works fine, ive got response from auth server:
[image]

dawidszyszka, 2023-07-29:
it is looking like problem with new users trying to login. @Mr-MIBonk is this site hosted on your server?

FeralMib, 2023-07-29:
> it is looking like problem with new users trying to login. 
Thanks for the error report, I handle the server codebase. I recently did a big upgrade, and yeah while I tested (and had to fix) the login plugins, in hindsight I didn't come up with a way to test new user signup.

I'll look into fixing it certainly, but probably take a couple of days sorry!

mikecitt, 2025-01-04:
@FeralMib what is going on with mibwiki website?