Mr-MIBonk/M.I.B._More-Incredible-Bash #298: No GEM Menu after update - MHI2Q

Julian881, 2023-07-29:
S5 - 2019 - MHI2Q_ER_AUG22_P5092 - MU 1329

Everything went ok with the update but when i try to open the GEM it goes directly to the red menu like the update didn't take place.
[backup.zip](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12207739/backup.zip)

Hoping for a solution.  Thank you!

Julian881, 2023-07-29:
[MU1329-JTB-07404.02.1911730571.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12207755/MU1329-JTB-07404.02.1911730571.log)

harman-f, 2023-07-30:
LOG file is incomplete.

Get new SD card and install again.
SD has to stay in SD1 to be able to use M.I.B in GEM.

Julian881, 2023-07-31:
Yes, i kept it in while trying. Getting a new card and trying again today. 

Will let you know of the outcome. Thank you!

Julian881, 2023-07-31:
[MU1329-JTB-07404.02.1911730571.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12213695/MU1329-JTB-07404.02.1911730571.log)

New card, same thing happened.

harman-f, 2023-07-31:
Strange, try 3.7.1 release

Julian881, 2023-07-31:
Tried one more time and  it worked. Thanks!

closed 2023-07-31 by Julian881

Julian881, 2023-07-31:
Now I can’t manage to update fec. I am trying the last version of ROWBest regards,Bogdan

Julian881, 2023-07-31:
Best regards,Bogdan

Share LOG again.