Mr-MIBonk/M.I.B._More-Incredible-Bash #304: I can´t see the app connect icon after doing the patch

ghassan1986, 2023-08-03:
Hi,

I tried to unlock carplay/Mirrorlink (App-connect) on this porsche for one friend but after everything was done, unfortunately I can`t see the app icon, i tried to activate the icon again using GEM but it didn`t help.

where I can find the logs to attach it here?

zzz08900, 2023-08-03:
logs should be in your MIB SD card, under the backup folder

ghassan1986, 2023-08-03:
thx, here are the logs as per the attched links.

https://easyupload.io/67vy3j

that would be very pleasant of you guys to check where is the issue.

Thank you so much!

harman-f, 2023-08-06:
icons will onl show up after a phone is connected and CP/AA is activated on that phone

ghassan1986, 2023-08-10:
I tried both Apple and Android with original USB-Cables and both are not working, the icon doesn`t show up at all, do you know what is the issue here and how it can be done?

Thx

harman-f, 2023-08-15:
You did not install the patch so far:

`Patch folder /net/mmx/fs/sda0/patches/MHI2_ER_POG11_P3299_MU0807_PATCH NOT found on SD card`

Please check instruction in Wiki again ;)

closed 2023-08-15 by harman-f

ghassan1986, 2023-08-15:
It was there, I put it and did the patch 2 times and it shows successful

ordekalo, 2024-11-17:
change permission of files (from read-only)

Maltman3000, 2025-05-02:
ghassan1986 can you send me the patch collection MHI2_ER_POG11_P3299 as mibsolution is down?