Mr-MIBonk/M.I.B._More-Incredible-Bash #305: carplay patch MHI2_ER_VWG13_P4521

FAOUZISSAM, 2023-08-03:
BONJOURS

JE SOUHAITE QUEL VERSION PATCH JE PEUT  METTRE SUR MHI2_ER_VWG13_P4521 POUR ACTIVER LA FONCTION CARPLAY ET COMMANDE VOCAL

CDT

Cupidstunt79, 2023-08-06:
I’d update to the MHI2_ER_VWG13_P4525 AIO firmware and everything you want will be done in one go

harman-f, 2023-08-06:
https://github.com/harman-f/MHI2_MIB2_AIO_FW_Update_Template/wiki/AIO-installation

closed 2023-08-06 by harman-f

FAOUZISSAM, 2023-08-07:
thank you for your answers but please where i can find the file to download it because it is not available

FAOUZISSAM, 2023-08-08:
MHI2_ER_VWG13_P4525 AIO version is not available
[image]