Mr-MIBonk/M.I.B._More-Incredible-Bash #306: Multiple Bugs Spotted MHI2_KR_AU57x_S0035 > MHI2_ER_AU57x_K3663

R3DcooDe, 2023-08-04:
Hello

Audi A6 2016 Korean with firmware MHI2_KR_AU57x_S0035 /  download mib latest version , copied to brand new fat 32 SD Card .   open red menu started update - engine off, ignition on  after first reboot a Green Screen Doing its magic appears and stuck there forever i waited over 15 minutes  nothing happened.  i proceed with Force reboot,    after unit boot up to normal mode there was no mib installed -  so i repeat step 1 again , and this time mib was Installed  successfully..    i opened mib and proceed with conversation change to EUROPE after mib finished i run EU firmware from red menu -  firmware started updating on Korean Language  until after first reboot unit switched back to English's.. 

Part 2 - Now Unit is with EU Firmware  MHI2_ER_AU57x_K3663

i opened mib and Patch AIO  everything was done successfully  BUT after reboot there is no AUDI SMARTPHONE INTERFACE so no Carplay Andorid Auto.     however maps was installed fine without any problem //   when i go to setup mmi audi smartphone interface is grayed out , gray color 

here are the backup from mib backup folder for investigating

Link: https://we.tl/t-ybphqgrL0K

Cupidstunt79, 2023-08-06:
Do you have a CarPlay enabled usb port, from what I’ve read not all are CarPlay enabled some and are just for phone charging

R3DcooDe, 2023-08-06:
> Do you have a CarPlay enabled usb port, from what I’ve read not all are CarPlay enabled some and are just for phone charging

yes mate it has 2 stanart usb ports like on all audi a6 2016 mhi2 - the thing is it does not show at menu, there is no audi smartphone activated :/

harman-f, 2023-08-06:
seems the finalscript did not run properly on the 1st run.

This can happen....

You did install the 3663 AIO?

R3DcooDe, 2023-08-07:
> seems the finalscript did not run properly on the 1st run.
> 
> This can happen....
> 
> You did install the 3663 AIO?

nope i didn't install 3663 AIO -   i Installed K3663 and after installed MIB Again and run patch from mib- i always do it in this way

closed 2023-08-15 by harman-f

je8ker, 2024-01-29:
How do install k3663? i don't install k3663

blvckbeard1337, 2024-04-20:
@R3DcooDe @harman-f  Can you describe me in short term how should i convert my system MHI2_KR_AU57x_S0035 to MHI2_ER_AU57x_K3663,

I normally download the new version but I could'nt update because It says it doesnt support region. If you can just give me an idea how to fix it.

hguri, 2024-04-20:
hello i am interested to do also this conversation from KR to EU , to my audi a5 with mib2 ... can you help please ?! i have vcp and dlink or can make also without it(vcp i have, but dlink not) ?! thanks

je8ker, 2024-04-25:
im SW version MHI2_KR_AU57x_S0035 install MIB but activate apple carplay but not activate carplay 
i want try SW update or EU SW install my car how to activate car play
im a6 c7 F/L model SW version MHI2_KR_AU57x_S0035

R3DcooDe, 2024-04-25:
hello 50 euro paypal - i can help you convert and install

dielli12, 2024-07-29:
Hello R3DcooDe,

I'm having similar issues regarding to the conversion of the MHI2 System from Korea to Europe. 
I will pay you to help me. For more info write to me at: dielli.b.lakna@gmail.com

rept1leuser, 2024-08-08:
Hi, can u please tell me where to find the update for the firmware update i need the excact one but cant seem to find

T3r3A, 2024-11-14:
@R3DcooDe Please can you tell how much did the EU firmware take to install ? As i see from other forums it takes 7 hours..
Thanks