Mr-MIBonk/M.I.B._More-Incredible-Bash #310: MHI2_JP_VWG11_S0021_MU0014_PATCH

nm-correia, 2023-08-10:
**Description**

Missing patch for 2016 VW Golf MK7 Japanese import (MHI2_JP_VWG11_S0021_MU0014).
I know that the FW is a bit old (not finding anywhere a newer version online to update) but was hoping to have some luck here 😄 
Not able to upload backup files at the moment (mibsolution.one is returning 502).

harman-f, 2023-08-15:
That one is really old ;)

JP FW updates are not public so far...

What do you want to patch?
CP/AA is most likely not supported by that FW.

nm-correia, 2023-08-16:
That was my guess 🙃

Yeah I was hoping to enable CarPlay and EU navigation. But without a fw update that's not possible right? If you have any other suggestion would really appreciate it 😄

harman-f, 2023-08-16:
Car is in EU now?

nm-correia, 2023-09-30:
Issue solved by installing directly MHI2_ER_VWG11_K3342_1_AIO_MU1427_20230318.7z with modified metainfo2.txt file.

closed 2023-09-30 by nm-correia

Jimmyb16b, 2024-06-20:
@nm-correia Hi, I have a japanese import VW golf with MHI2_JP_VWG11_S0030. I want to change the firmware to the one you have listed above. Did you just do a software update in the developer menu, or did you do a region change in the GEM menu? Also could you share some information about the modified metainfo2.txt file? was this just to get maps working? Thanks!

nm-correia, 2024-06-23:
@Jimmyb16b I have done a fw update to MHI2_ER_VWG11_K3342_1_AIO_MU1427_20230318 and after that I have trigger the region change through the GEM menu. I used the modified metainfo2.txt from here: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/issues/291 (I had no issues with loading time after the fw upgrade)

Grant2468, 2025-07-30:
@nm-correia Hey mate, Im trying to achieve the same conversion. How did you go about installing the FW MHI2_ER_VWG11_K3342_1_AIO_MU1427_20230318 Did you have to run the eu conversion in M.I.B first? As when I run that conversion script I get errors saying my firmware is too old. Im starting on MHI2_JP_VWG11_P0030 btw