Mr-MIBonk/M.I.B._More-Incredible-Bash #314: STRANGE BUG !! with Pre-V3.7.1 - MHIG Edition - AUDI A6 2016 MHI2_xxx Korea

R3DcooDe, 2023-08-16:
Hello 

I Spotted an ANOMALY during converting 2 Audi A6 2016 with korean firmware  a) MHI2_KR_AUG11_P0009 // b) MHI2_KR_AU57x_S0035

i used latest MIB 3.7.1 - mib installed smoothly and opened from green menu. - i did procedures like always - i convert to europe / force restart unit and run conversion again from mib ...    after I put European firmware MHI2_ER_AU57x_ K3663 --  

i open red menu - update - sd1 - user defined- select all -- after restart unit start updating on korean language ( not a big deal ) - unit update in korean language only with mib version from qr edition and v3.7.1  --- with older mib update starts in english //))  

now it come the fun part --  after updating part 4 -- comes UPDATE 5/13  --- 

KBD_FC7/TPA/App --  i measured with phone it take excatly 56 seconds for 1% to complete -  so to go 100% it takes 100 minutes ( updating system only update 5 takes ages to update and only in audi a6 korean cars ONLY A6 2016 - KOREA ;; ALL OTHER REGION MODELS ARE FINE THEY UPDATE NORMALY ))) 

now the problem :::  WHEN DOING EU CONVERT WITH NEW MIB VERSION V3.7.1 this update 5/13 --- 
KBD_FC7/TPA/App & KBD_FC7/TPA/Bolo all devices inside this folder 5 kbd -- goes 100% it DOES NOT go to update step6 and so on,  but it KEEP UPDATING STEP 5 over and over and over stuck in loop..   so i left one car 7 hours just to wait and see what happens - nothing step 5 just stuck in loop doing update to 100% and start over again --- ( by the way i found how to deal with this issue and finish the update succesfully at end :D ))  

now fun part it is only with new mib version it happens this ERROR / bug 

i tested mib 3.5 spring edition -  it works well - so update start in english and when part 5 comes // KBD_FC7/TPA/App updating 1% goes 1 minute - so 100% = 100 minutes but when this go to 100% it finish and start updating step 6 and so on 

im curios what step 5 update what are those KBD why only on audi A6 2016 Korea it takes 4ever to update only part 5 

here are backup from mib -- to investigate further ( also here is a very old korean firmware : ( MHI2_KR_AUG11_P0009 ) to add support in future :) 

LINK : https://we.tl/t-HhdaaOCU8J

[image]

[image]

artt05, 2024-03-05:
Hello, i am trying to convert HMI2_KR_AUG11_P0009 to european model for 2016 audi a6 but an error is occurring “Variant conflict” do you think you can help me ?

R3DcooDe, 2024-03-05:
yes mate definitely i can help you step by step

artt05, 2024-03-20:
Do you have any social media i can dm you?

hguri, 2024-04-20:
hello i am interested to do also this conversation from KR to EU , to my audi a5 with mib2 ... can you help please ?! i want vcp and dlink or can make also without it(vcp i have, but dlink not) ?! thanks

D4612, 2024-08-24:
Hello R3D cooDe,

I texted back to you via e-mail regarding to this. I didn't see your email because it was sent to me as spam, so you can check and we can keep the conversation going.

Best regards,
D461.

R3DcooDe, 2024-08-24:
hello mate, can you tell me where the issues was,  what do you need

D4612, 2024-08-24:
The issue is when you need to update firmware after m.i.b procedure, do you
need to select all components or select all but leave kbd_fc7 to N, because
I also heard it's not needed. And is it really important what m.i.b version
you use for conversion?
Best regards,
D461.

D4612, 2024-08-26:
Waiting for your kind response....

R3DcooDe, 2024-08-26:
where do you live